Ameba-AIoT/ameba-arduino-d #1: ameba_d_tools, please support MacOS and Linux

Pillar1989, 2020-05-04:
I believe that 80% of individual developers are using MacOS and Linux. Please provide as soon as possible. We really don't want to use Windows at home.

technoblogy, 2020-06-04:
I'm on MacOS and would also be interested in exploring the Ameba boards with the Arduino IDE.

ambiot, 2020-06-05:
I am working on supporting both MacOS and Linux. The tools will be released soon.

technoblogy, 2020-06-05:
Great! I'll be happy to beta test it and provide feedback - what board would you recommend?

closed 2020-07-01 by Pillar1989

technoblogy, 2020-07-01:
Why have you closed this? Is the MacOS support now available? What about my question:

> Great! I'll be happy to beta test it and provide feedback - what board would you recommend?

Pillar1989, 2020-07-01:
It's been tested. Perfect support.You can try it. @technoblogy

technoblogy, 2020-07-01:
I get:

    Error downloading https://github.com/ambiot/ambd_arduino/raw/master/Arduino_package/package_realtek.com_amebad_index.js

Pillar1989, 2020-07-01:
Looks like GitHub has a grudge against you and won't let you download it.

technoblogy, 2020-07-01:
Sorry, my bad - I had left the "on" off the end of .json because in my browser it had wrapped onto the next line. Works now.