Ameba-AIoT/ameba-arduino-d #2: Problems uploading from the Arduino IDE

technoblogy, 2020-07-22:
Today I received an Ameba RTL8722DM IoT Development Board from Seeed Studio.

When I connect the **CON3** USB port to my computer I get **/dev/cu.usbserial-AV0JH7JH** on the **Port** menu.

Selecting this and uploading the Blink sketch then gives:

```
An error occurred while uploading the sketch
cp tools/macos/image_tool/imgtool_flashloader_amebad.bin ./
Please enter the upload mode (wait 5s)
    05
    04
    03
    02
    01
./tools/macos/image_tool/amebad_image_tool /dev/cu.usbserial-AV0JH7JH
```
Please let me know what I need to do to get this to work.

technoblogy, 2020-08-06:
I've upgraded to the Boards file 3.0.4 and I now seem to be able to upload a file with the message:

    Done uploading.

but I can't get any of the examples to run. For example, **Blink** doesn't blink an LED, and **SimpleHttpExample** doesn't print anything to the Arduino Serial Monitor.

Please can you suggest one example that will prove that my Ameba RTL8722DM IoT Development Board is working.

M-ichae-l, 2020-08-27:
@technoblogy 
Looks weird. Could you show the full log from the Arduino IDE?
One thing I experienced is that do not do anything when the red LED light-up during the image uploading. Otherwise, the image uploading will fail.

technoblogy, 2020-08-28:
I don't get a red LED during upload. Here's the output:

````
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/david/Library/Arduino15/packages -hardware /Users/david/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/david/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/david/Documents/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=0403_6001 -ide-version=10813 -build-path /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_154052 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_tools.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.0.3 -prefs=runtime.tools.ameba_d_tools-1.0.3.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.0.3 -verbose /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_modified_sketch_743883/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/david/Library/Arduino15/packages -hardware /Users/david/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/david/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/david/Documents/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=0403_6001 -ide-version=10813 -build-path /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_154052 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_tools.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.0.3 -prefs=runtime.tools.ameba_d_tools-1.0.3.path=/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.0.3 -verbose /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_modified_sketch_743883/Blink.ino
Using board 'ameba_rtl8721d' from platform in folder: /Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4
Using core 'arduino' from platform in folder: /Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4
Detecting libraries used...
/Users/david/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/bin/arm-none-eabi-g++ -ffunction-sections -march=armv8-m.main+dsp -mthumb -mcmse -mfloat-abi=hard -mfpu=fpv5-sp-d16 -g -gdwarf-3 -nostartfiles -nodefaultlibs -nostdlib -O2 -D__FPU_PRESENT -gdwarf-3 -fstack-usage -fdata-sections -nostartfiles -nostdlib -Wall -Wpointer-arith -Wundef -Wno-write-strings -Wno-maybe-uninitialized -c -Wextra -w -x c++ -E -CC -mcpu=cortex-m33 -DF_CPU=200000000L -DARDUINO=10813 -DARDUINO_AMEBA -DARDUINO_ARCH_AMEBAD -mthumb -DBOARD_RTL8721D -DUSB_VID= -DUSB_PID= -DUSBCON "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"RTL8722DM/RTL8722CSM\"" -Wl,--start-group -I. -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/project/realtek_amebaD_va0_example/inc/inc_hp -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/os/freertos -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/include -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/portable/GCC/ARM_CM33/non_secure -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/os/os_dep/include -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/soc/realtek/amebad/misc -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/common/api/network/include -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/common/api -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/common/api/at_cmd -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/common/api/platform -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/common/api/wifi -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src/crypto -I/Users/david/Library/Arduino15/packages/realtek/hardware/AmebaD/3.0.4/system/lib [... cut ...]
I uploaded the Blink example, and the LED doesn't flash after upload so I assume the upload has failed.

Do I need to press RESET or UART DOWNLOAD?

If I open Serial Monitor at 115200 baud and press RESET I get:

````
#calibration_ok:[2:19:11] 
#interface 0 is initialized
interface 1 is initialized

Initializing WIFI ...
RTL8721D[Driver]: The driver is for MP
[FAST_CONNECT] Fast connect profile is empty, abort fast connection

WIFI initialized

init_thread(54), Available heap 0x29080
````
Thanks,
David

ambiot, 2020-08-31:
@technoblogy 
Looks like you did not enter the upload mode. 
Did you follow the instruction by the link [https://www.amebaiot.com/en/amebad-arduino-getting-started/](url) ?

> The Arduino IDE will compile first then upload. During the uploading process, users are required to enter the upload mode of the board. Arduino IDE will wait 5s for DEV board to enter the upload mode.

> To enter the upload mode, first press and hold the UART_DOWNLOAD button, then press the RESET button. If success, you should see a green LED flashing on the DEV board.

technoblogy, 2020-08-31:
OK, thanks, I missed that. It's working now.

closed 2020-08-31 by ambiot