Ameba-AIoT/ameba-arduino-d #3: BLE HID keyboard client example

cheatplayer, 2020-08-26:
I am trying to get input from a bluetooth keyboard.
Scan and connect successed,get service look like ok, but can't read input value characteristics.
Here is my code.
```
#include "BLEDevice.h"
#include "BLEHIDClient.h"

#ifdef __cplusplus
extern "C" {
#endif

#include "simple_ble_client.h"
#include "profile_client.h"
#ifdef __cplusplus
}
#endif

T_SERVER_ID BLEHIDClient::_hidClientId=0;
void (*BLEHIDClient::_pDefaultCB)()=nullptr;

BLEHIDClient::BLEHIDClient(){

}

void BLEHIDClient::addClient(uint8_t max_link_num){
    client_init(1);
    _hidClientId=simp_ble_add_client(clientCallback,max_link_num);
    // client_register_general_client_cb(clientCallback);
}

void BLEHIDClient::enableNotify(uint8_t conn_id){
    simp_ble_client_set_v3_notify(conn_id,1);
}

uint8_t BLEHIDClient::discoveryHIDServices(uint8_t conn_id){
    return (client_by_uuid_srv_discovery(conn_id,_hidClientId,0x1812));
}

void BLEHIDClient::setDefaultCallback(void (*fCallback)()){
    _pDefaultCB=fCallback;
}

uint8_t BLEHIDClient::readHIDKeyboard(uint8_t conn_id){
    return (client_attr_read(conn_id,_hidClientId,0x2A22));
}

uint8_t BLEHIDClient::read2902(uint8_t conn_id){
    return (client_attr_read(conn_id,_hidClientId,0x2902));
}

T_APP_RESULT BLEHIDClient::clientCallback(T_CLIENT_ID client_id, uint8_t conn_id, void *p_data){
    T_APP_RESULT app_result = APP_RESULT_SUCCESS;
    _pDefaultCB();
    if(client_id==_hidClientId){
        T_SIMP_CLIENT_CB_DATA *p_simp_cb_data = (T_SIMP_CLIENT_CB_DATA *)p_data;
        switch(p_simp_cb_data->cb_type){
            case SIMP_CLIENT_CB_TYPE_READ_RESULT:{
                Serial.println("[1.cb_type] read");
                switch(p_simp_cb_data->cb_content.read_result.type){
                    case SIMP_READ_V1_READ:{
                        Serial.println("[2.cb_content] read");
                        Serial.print("cause ");
                        Serial.println(p_simp_cb_data->cb_content.read_result.cause);
                        Serial.print("data notify ");
                        Serial.println((int)p_simp_cb_data->cb_content.read_result.data.v3_notify_cccd);
                        Serial.print("data v4_indicate_cccd ");
                        Serial.println((int)p_simp_cb_data->cb_content.read_result.data.v4_indicate_cccd);
                        Serial.print("data v1_read size ");
                        Serial.println(p_simp_cb_data->cb_content.read_result.data.v1_read.value_size);
                        break;
                    }case SIMP_READ_V3_NOTIFY_CCCD:{
                        Serial.println("[2.cb_content] notify");
                        break;
                    }case SIMP_READ_V4_INDICATE_CCCD:{
                        Serial.println("[2.cb_content] indicate");
                        break;
                    }default:{
                        Serial.println("[2.cb_content] other");
                        break;
                    }
                }
                break;
            }
            case SIMP_CLIENT_CB_TYPE_NOTIF_IND_RESULT:{
                Serial.println("[1.cb_type] notify or indicate");
                break;
            }
            default:{
                Serial.println("[1.cb_type] other");
                break;
            }
        }
    }
    
    return app_result;

}
```

When I call `readHIDKeyboard` the value size in callback always 0.
DId I use the right api? And how to enable notification from host device?

Please give more infomation. Thx.

M-ichae-l, 2020-08-27:
@cheatplayer 
Let me review it first.

Aurical, 2020-08-28:
Specifically on why the value size callback is zero, the third argument of client_attr_read is supposed to be a handle, not a UUID. the handle is similar to an address supplied by the server, used to refer to an attribute, and is obtained during the characteristic discovery process, which in this case will be saved in the simple_ble_client handle table.

More generally, I think the approach is wrong in using simple_ble_client, since the simple_ble_client will receive the discovery state callback and discovery result callback, and it will be looking for attributes corresponding to a simple_ble_service structure instead of a ble hid keyboard structure.

cheatplayer, 2020-08-28:
@Aurical Thank you for your explanation of handle. 
I am wrong in using simple_ble_client, but I am not find hid_ble_client to use.
If I want implement this function is there any low level api exposed?

implement |  
-- | --

implement |  
-- | --

implement |  
-- | --

Aurical, 2020-08-28:
@cheatplayer my approach would be to use API in profile_client.h to write a hid_ble_client, while referring to existing client source code for the formatting and overall structure.

cheatplayer, 2020-08-28:
@Aurical thanks! I will try it.

ambiot, 2020-09-03:
@Aurical  @cheatplayer 
The issue is closed as there is no further issue.

closed 2020-09-03 by ambiot