Ameba-AIoT/ameba-arduino-d #4: Serial fixed at 115200 baud

technoblogy, 2020-08-31:
In the following program:

````
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Hello");
  delay(1000);
}
````
the statement `Serial.begin(9600)` seems to be ignored, and the baud rate is left at 115200.

M-ichae-l, 2020-08-31:
@technoblogy 
The default serial baud rate of RTL8722 is set as 115200. And we do not recommend users to change it. So in the SDK we use a marco to lock the serial baud rate. 
You can change it in the 

> C:\Users\username\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\packagenumber\cores\arduino\LOGUARTClass.cpp

Set `#define LOG_UART_MODIFIABLE_BAUD_RATE 0` as `#define LOG_UART_MODIFIABLE_BAUD_RATE 1`

technoblogy, 2020-08-31:
Thanks - I can live with it set to 115200 - just wanted to check.
David

closed 2020-08-31 by technoblogy