Ameba-AIoT/ameba-arduino-d #11: Problems Creating a Wi-Fi Soft Access Point

technoblogy, 2020-09-14:
I'm having a lot of problems creating a Wi-Fi soft access point using a call to **WiFi.apbegin(ssid, pass, channel)**.

If I run the **WiFiAPMode** example it works OK, and I can join the network **yourNetwork** from my computer with the password **Password**.

However, if I try to integrate the code into my program I get the message "AP mode already started", and I can see the network in the list of networks, but if I try to connect to the network I get a message saying the password is not a WPA2 password.

The differences between my program and the example are:

- I'm not calling **printWifiData()** or **printCurrentNet()**.
- I've changed **loop()** to an empty loop; in other words:

      void loop() { }

Any suggestions?

Aurical, 2020-09-15:
This code seems to work for me, can you try it?

```
#include <WiFi.h>

char ssid[] = "yourNetwork";  //Set the AP's SSID
char pass[] = "Password";     //Set the AP's password
char channel[] = "1";         //Set the AP's channel
int status = WL_IDLE_STATUS;  // the Wifi radio's status

void setup() {
    //Initialize serial:
    Serial.begin(115200);
    // Check wifi status
    status = WiFi.status();
    // attempt to start AP:
    while (status != WL_CONNECTED) {
        Serial.print("Attempting to start AP with SSID: ");
        Serial.println(ssid);
        status = WiFi.apbegin(ssid, pass, channel);
        delay(10000);
    }

    //AP started:
    Serial.println("AP mode started");
}

void loop() {
    // check the network connection once every 10 seconds:
    delay(10000);
}
```

technoblogy, 2020-09-15:
Yes, it works, but doesn't work if I remove the line:

    delay(10000);

In my actual application the main loop will be executing other code. Is there a call I need to make periodically to check the network connection?

Thanks, David

Aurical, 2020-09-15:
It looks like the RTOS running in the background does not like a empty loop. using `vTaskDelay(1 / portTICK_PERIOD_MS);` is sufficient to keep it running.

technoblogy, 2020-09-16:
That works - thanks!

technoblogy, 2020-09-17:
Now I'm getting a new problem. When I try and create a Soft Access Point I get the error:

    Error!!!upd_new error

but the Soft Access Point seems to have succeeded, and I can see it in my list of Networks.

However, when I try to connect to it I get a series of errors:

    Cannot allocate pbuf to receive packet

What's the problem here? It was working at first. Thanks, David

Aurical, 2020-09-17:
Do you mind posting your code here? Need to get an idea of what it is doing in the main loop.

technoblogy, 2020-09-17:
I'll try and create a cut-down version that shows the problem.

technoblogy, 2020-09-17:
OK, I think I've figured it out. I omitted the line:

    status = WiFi.status();

It seems it's necessary to call WiFi.status() before calling WiFi.apbegin().

Thanks, David

technoblogy, 2020-09-19:
I've now got a version of uLisp, my Lisp for microcontrollers, available for the Ameba RTL8722DM IoT Development Board:

http://www.ulisp.com/show?39RE

Is there somewhere more appropriate I could mention this?

Aurical, 2020-09-19:
It looks like Realtek developers do check here occasionally. Alternatively, you could try posting in the community forum at https://forum.amebaiot.com/

closed 2020-10-28 by ambiot