Ameba-AIoT/ameba-arduino-d #12: Creating a Wi-Fi Soft Access Point options

technoblogy, 2020-09-15:
A couple of other questions:

- Is there a call to disconnect a Soft Access Point. On the ESP8266/32 it's:

      WiFi.softAPdisconnect(true);

- Is there a call to create a hidden Soft Access Point? On the ESP8266/32 there's an optional fourth parameter:

      bool hidden = true;
      WiFi.softAP(ssid, pass, channel, hidden);

I've had a look in the source but can't find anything.
Thanks, David

Aurical, 2020-09-15:
- `WiFi.disconnect()` should disable the AP mode

- It looks like hidden SSID ap mode is not exposed in the Arduino facing API by default. The functionality is present in `wifi_conf.h` file, with the `wifi_start_ap_with_hidden_ssid` function. It should be possible to link this function into `wifi_drv.cpp` and implement hidden SSID ap mode in `WiFI.cpp`

ambiot, 2020-10-28:
The latest version v3.0.6 has added the hidden SSID function.

closed 2020-10-28 by ambiot