Ameba-AIoT/ameba-arduino-d #13: Switching between Central an Peripheral

BelikovVictor, 2020-12-15:
I catch permanent freeze after switch between Central to Peripheral. Call of BLE.beginPeripheral() blocks program permanently.

void WifiLoop()
{
  int connecting_flag = 0;
  // attempt to connect to Wifi network
  if (WiFi.status() != WL_CONNECTED)
  {
     Serial.println("WiFi disconnected");
    connecting_flag = 1;
    BLE.configScan()->stopScan();
    BLE.end();
    Serial.println("Begin per...");
    BLE.beginPeripheral();
    Serial.println("Start adv...");
    BLE.configAdvert()->startAdv();
  }

  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("Wait WiFi config..");
    // wait for connection
    delay(5000);
  }
  if (connecting_flag)
  {
    connecting_flag = 0;
    BLE.beginCentral(0);
    BLE.configScan()->startScan();
  }
}

void BleAdvSetup()
{
  //ble conf service
  BLE.configServer(1);
  configService.addService();
  configService.begin();

  // Wifi config service requires a specific advertisement format to be recognised by the app
  // The advertisement needs the local BT address, which can only be obtained after starting peripheral mode
  // Thus, we stop advertising to update the advert data, wait for advertising to stop, then restart advertising with new data

  BLE.configAdvert()->stopAdv();
  BLE.configAdvert()->setAdvData(configService.advData());
  BLE.configAdvert()->updateAdvertParams();
  delay(100);
  //BLE.beginPeripheral();
  //BLE.configAdvert()->startAdv();
}

void BleSetup()
{
  BLE.init();
  BLE.configScan()->setScanMode(GAP_SCAN_MODE_PASSIVE);    // Active mode requests for scan response packets
  BLE.configScan()->setScanInterval(5000);   // Start a scan every 500ms
  BLE.configScan()->setScanWindow(1000);     // Each scan lasts for 250ms
  BLE.configScan()->updateScanParams();
  BLE.configScan()->setScanDuplicateFilter(0);
  // Provide a callback function to process scan data.
  // If no function is provided, default BLEScan::printScanInfo is used
  BLE.setScanCallback(BleScanFunction);

  BleAdvSetup();

  BLE.beginCentral(0);
  BLE.configScan()->startScan();    // Repeat scans for 5 seconds, then stop
}

Aurical, 2020-12-15:
Can you provide more specifics?
from what I can see, I am guessing that it is occuring in the WifiLoop function, it might be caused by not ending the previous mode before switching between Central and Peripheral

BelikovVictor, 2020-12-15:
I call BLE.end() before switching. I attach the whole sketch.

[RL8722DMBridgeTest.zip](https://github.com/ambiot/ambd_arduino/files/5695712/RL8722DMBridgeTest.zip)

Aurical, 2020-12-16:
[fix.zip](https://github.com/ambiot/ambd_arduino/files/5700270/fix.zip)

you are right, there was something blocking the peripheral mode.
attached are some new files that should fix the issue, make a copy of the old files, and replace with the fixed files in the libraries folder.

a few points I noticed regarding the rest of the code:
1. advertising data will be lost after calling BLE.end(), so you will need to reconfigure advertising data after switching to peripheral mode
2. after connecting the board to wifi, the smartphone app used for WiFi config will remain connected until the user accepts the settings or quit the app, so it is not possible for the board to switch back to central mode immediately after connecting to wifi. Another criteria should be used to determine when to stop peripheral mode and switch to central mode, perhaps check if the board is still connected to a device before calling BLE.end and switching modes.

BelikovVictor, 2020-12-16:
Thank you for cooperation. I also want to note that the BLE.end() needs internal blocking and synchronization. Now we have to add delays when switching from scan mode to peripheral:
`BLE.configScan()->stopScan();`
`delay(5000);`
`BLE.end();`

Aurical, 2020-12-16:
can you clarify on the internal blocking and synchronization?
do you mean BLE.end() should stop scans when called?

BelikovVictor, 2020-12-16:
I meant that BLE.end () should wait for the completion of the stop scan operation, or stopScan () should do it. Anyway, without delay (5000) this code does not work, this is not the correct behavior

Aurical, 2020-12-17:
I have done some testing and could not replicate the behavior. From what I can see, the code still works without the delay(5000).

Here is the code snippet:
```
  if (WiFi.status() != WL_CONNECTED)
  {
    Serial.println("WiFi disconnected");
    connecting_flag = 1;
    Serial.println("Stop scan");
    BLE.configScan()->stopScan();
    BLE.end();
    Serial.println("BLE end");
    Serial.println("Begin peripheral");
    BLE.beginPeripheral();
    Serial.println("Start adv");
    BLE.configAdvert()->startAdv();
  }
```

and here is the log output:
```
15:45:47.136 -> [BLE Device] GAP scan start
15:45:47.136 -> Scan Data 1
15:45:47.136 -> Scan Data 2
15:45:47.136 -> Scan Data 3
15:45:47.182 -> Scan Data 4
15:45:47.228 -> WiFi disconnected
15:45:47.228 -> Stop scan
15:45:47.228 -> BLE end
15:45:47.228 -> Begin peripheral
15:45:47.228 -> Adv update
15:45:47.228 -> GAP cb reg
15:45:47.228 -> Task create
15:45:47.228 -> Coex init
15:45:47.228 -> BT stack start
15:45:47.228 -> gapMsgHandlerDefault: subtype 1
15:45:47.228 -> devStateEvtHandlerPeriphDefault: init state 1, adv state 0, cause 0x0
15:45:47.228 -> Coex on
15:45:47.321 -> Start adv
```

BelikovVictor, 2020-12-17:
Indeed, apparently I ran the code before the fix.
Yes, everything works without delay.

Aurical, 2020-12-17:
That is good to hear, thanks for the your feedback

closed 2021-02-03 by ambiot