Ameba-AIoT/ameba-arduino-d #14: Wi-Fi scan results with BSSID of AP

edward17829991, 2021-02-02:
Hi 

Can I get the BSSID of the scanned APs?
I need to identify and connect to hidden Wi-Fi through the scanned hidden network.
I've tried to implement it myself, but the BSSID with the first byte equals 0x00 will lead to all zeros in consequent BSSID bytes.
Maybe some bugs in the precompiled binary object files?

I also notice there's a hardcoded limit of 64 scanned APs limit in the binaries, hope it will be available for configuration.

Thanks

xidameng, 2021-02-03:
> Hi
>
> Can I get the BSSID of the scanned APs?
> I need to identify and connect to hidden Wi-Fi through the scanned hidden network.
> I've tried to implement it myself, but the BSSID with the first byte [...]

Hi printing out BSSID is supported in the current version of the arduino SDK, please check out the "WiFi.cpp" file under the realtek arduino package, alternatively, its available here as well https://github.com/ambiot/ambd_arduino/blob/47cd35c7935c32e57287757e30781943d57c33b3/Arduino_package/hardware/libraries/WiFi/src/WiFi.cpp#L123

edward17829991, 2021-02-03:
Hi xidameng,

Thanks for your quick response.
As to the function you referred to, if I understand it correctly, it gets the BSSID of the *currently connected* AP, but not the BSSIDs of all scanned APs (not yet connected). 
In other words, for acquiring the SSID and RSSI of the scanned APs (not yet connected), there are functions as following:
https://github.com/ambiot/ambd_arduino/blob/47cd35c7935c32e57287757e30781943d57c33b3/Arduino_package/hardware/libraries/WiFi/src/WiFi.cpp#L157
https://github.com/ambiot/ambd_arduino/blob/47cd35c7935c32e57287757e30781943d57c33b3/Arduino_package/hardware/libraries/WiFi/src/WiFi.cpp#L162
But no similar function for BSSID.

Hope I've clarified my issue.

p.s. 
I've tried to extend a function:
`uint8_t* BSSID(uint8_t networkItem);`
with logic very similar to the one of
`char* SSID(uint8_t networkItem);`,
which works OK for now.

Best, 
Edward

xidameng, 2021-02-03:
Glad you find the way to get BSSID, if it works fine, consider pull request to the repo so others with similiar need will also benefit from your work :)

closed 2021-02-03 by ambiot