Ameba-AIoT/ameba-arduino-d #15: is it compatible with RTL8720DN?

bkrajendra, 2021-03-01:
is it compatible with RTL8720DN mocule?

xidameng, 2021-03-01:
Check out this repo for RTL8720DN
https://github.com/xidameng/ambd_arduino

bkrajendra, 2021-03-01:
Thank you for your quick reply. I'll look into it.

bkrajendra, 2021-03-01:
I have also found this repo with more ported libraries and information for RTL8720DN
https://github.com/mikey60/BW16-RTL8720DN-Module-Arduino

xidameng, 2021-03-02:
> I have also found this repo with more example sketches and information for RTL8720DN
> https://github.com/mikey60/BW16-RTL8720DN-Module-Arduino

Thanks for the info~ Hope you find the RTL8720DN handy to use, if you do enjoy, you may consider fork the repo and contribute yourself~

bkrajendra, 2021-03-02:
Yes definitely!

closed 2021-03-04 by ambiot