Ameba-AIoT/ameba-arduino-d #16: OTA Updates

mikey60, 2021-03-17:
Is there a way to do Over-The-Air (OTA) updates using WiFi?

xidameng, 2021-03-17:
Hi, this feature has not yet supported on Arduino IDE, but if you are interested, you may check out the ambd_sdk repo under this same account and the OTA example over there at https://github.com/ambiot/ambd_sdk/tree/master/component/common/example/ota_http

ambiot, 2021-04-08:
Plan to support this function in the future after we release V3.0.8

jdt1, 2023-02-02:
> Plan to support this function in the next release.

Is there any progress on this function?

M-ichae-l, 2023-02-06:
@jdt1  it is ongoing. Windows is ready, we are working on linux and macos.

jdt1, 2023-02-06:
> @jdt1 it is ongoing. Windows is ready, we are working on linux and macos.

If Windows is ready, can I download it already?

M-ichae-l, 2023-02-07:
> 

the PR is ready but has conflicts with master. I will inform the PR holder to update. Then apply to merge.

S10143806H, 2023-02-20:
> > @jdt1 it is ongoing. Windows is ready, we are working on linux and macos.
> 
> If Windows is ready, can I download it already?

Hi @jdt1,

Thanks for your waiting. 

The current OTA tool is compatible with the latest AmebaD Arduino SDK. However, I am still working on the tool executable for macOS and Linux.  

I will update the PR once everything is done. Meanwhile, all the conflicts will be solved.

M-ichae-l, 2023-04-12:
Closed as the feature already supported.

closed 2023-04-12 by M-ichae-l