Ameba-AIoT/ameba-arduino-d #31: #include <map> won't compile

FL0WL0W, 2021-07-06:
It seems like everytime i want to include a standard library, the project fails to compile. usually due to naming conflicts between things in the SDK and things in the Arduino Core.

For instance if you #include <list> it will break do to the min/max methods in both arduinocore and the sdk.

This is what i get when trying to include <map>
 ```
In file included from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/ext/string_conversions.h:43:0,
                 from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/bits/basic_string.h:5429,
                 from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/string:52,
                 from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/stdexcept:39,
                 from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/array:39,
                 from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/tuple:39,
                 from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/bits/stl_map.h:63,
                 from /home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/map:61,
                 from /tmp/arduino_modified_sketch_773068/WiFiAPMode.ino:2:
/home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/cstdio:127:11: error: '::printf' has not been declared
   using ::printf;
           ^~~~~~
/home/daniel/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/cstdio:137:11: error: '::sprintf' has not been declared
   using ::sprintf;
           ^~~~~~~
```

I appreciate the work that has been done to try and include these useful chips into the Arduino environment. However, This still needs a lot of work and a lot of testing to make sure users have a good experience like they do with the similar ESP8266.

xidameng, 2021-07-06:
Hi Daniel,

Thanks for your feedback, please check the following for the fix,

## Steps
1. Uncomment the code on the following line
https://github.com/ambiot/ambd_arduino/blob/c8426ff8331b27e85c7f3e785ea5e6bae5ae7bc5/Arduino_package/hardware/cores/arduino/Arduino.h#L28

2. Add the following code before you `#include "map"`
```c
#undef max
#undef min
```

## Last but not the least
These error occurs because Ameba has its own set of standard library implementation which conflicts with ones used in the C++ std lib, and ameba's own std lib has significantly smaller code size, thus reducing the RAM usage. These issue also exist with libraries used by ESP8266, but were probably fixed by the community, so we also need community's help to spot these issues so as to increase everybody's user experience.

FL0WL0W, 2021-07-06:
Thank you so much for the quick response! Those fixes worked! I assume these will be implemented in the main branch later on and the issue will be closed out?

I can tell there has been a lot of effort to provide examples, and I look forward to opening some more issues to further polish the Arduino implementation.

closed 2021-07-07 by ambiot