Ameba-AIoT/ameba-arduino-d #32: Add support for RTL8720 BW16 board variant

FL0WL0W, 2021-07-06:
Please reference https://github.com/mikey60/BW16-RTL8720DN-Module-Arduino

This should be relatively straight forward to implement, just add another variant and update the variant.cpp

xidameng, 2021-07-07:
Hi Daniel, thanks for your feature request, we are glad to share with you that our community contributor @happyme531  has already ported `ambd_arduino` to RTL8720DN and we have also already added the support to this chip in our Arduino SDK, look no further than this,

https://github.com/ambiot/ambd_arduino/tree/master/Arduino_package/hardware/variants/rtl8720dn_bw16

Please be sure to use the link below to update your arduino 3rd party package and install the latest version of ameba package to have BW16 installed on your Arduino IDE!

https://github.com/ambiot/ambd_arduino/raw/master/Arduino_package/package_realtek.com_amebad_index.json

FL0WL0W, 2021-07-07:
Looks like it was just added today! I redownloaded 3.0.8 and that variant is now there! Great to see such great support from the ambiot team!

closed 2021-07-07 by FL0WL0W