Ameba-AIoT/ameba-arduino-d #33: Had to update firmware before arduino sketch would work

FL0WL0W, 2021-07-06:
When I first got started with this board, When I would upload my sketch it would not update. I had to follow this guide along with modifying their python script to use my RS232 USB.  https://wiki.seeedstudio.com/Wio-Terminal-Network-Overview/

please add a way to update the firmware from the Arduino IDE like they have for burning the bootloader to the Atmel devices. it looks like there are firmware binaries in the tools directory, but no way to upload them from the arduino IDE, and i found no guide for how to update them other than from Seeedstudio

xidameng, 2021-07-07:
Hi Daniel,

My guess is that you are using RTL8720DN(BW16) module/dev board from B&T for your project right?

Please note that 

1. BW16 module is designed and made by B&T, thus it is not the official `Ameba Hardware` from Realtek (except the IC), thus we can't guarantee its performance
2. This issue has been reported on our [Ameba Forum](https://forum.amebaiot.com/t/how-to-load-bw16-program-with-arduino/517), and our community members have already fixed it and explained the reasons behind, you may click the link above to learn more.

xidameng, 2021-07-07:
You may also consider to join our forum there to discuss more with our friendly ameba developers :)

FL0WL0W, 2021-07-07:
Hi Xidameng, I will join the forum. I understand that this is the default B&T firmware that it is shipped with. It looks like all of the tools and even the binarys are included in the tools directory of arduino. I think there is a way to include a shortcut under the tools drop down in arduino for the image_tool.exe gui. This would be a fantastic addition as that is where i looked first to see if there was a firmware update that needed to be loaded. The arduino directory is kind of a pain to access since it is in a hidden folder and most new users will not look there. Thank you for all of your help!

xidameng, 2021-07-07:
> Hi Xidameng, I will join the forum. I understand that this is the default B&T firmware that it is shipped with. It looks like all of the tools and even the binarys are included in the tools [...]

Extension to Arduino IDE has to be created using Java instead of C, which could be an issue here. Also Arduino is moving towards IDE 2.0 which is based on eclipse's core, thus all java extension might not work by then. so to play safe, we will upload a `getting started` page as well as a `troubleshooting guide` for RTL8720DN module and development board soon, so if you have something to share with everybody using this module, you can also contribute by adding your own original content to our [readthedocs online documentation](https://amebaiotdocuments-zh-tw.readthedocs.io/en/latest/index.html)

ambiot, 2021-07-16:
Proceed to close.

closed 2021-07-16 by ambiot

brcisna, 2021-07-29:
FL0WL0W 
Am experiecing the same situation with two of the B&T rtlduino BW16 modules. The erasing procedure can not be done with a simple usb connected to the rtlduino,,,i am assuming? Am really struggling with this module. Have spent a LOT of time,,even trying how to figure out how to get these modules,,into upload mode. Now have this down to a sceince. I am assuming the only way to earse a factory burned rtlduino is with a usb to ttl adapter,,,which I have. Just now found this article explaining to erase the factory firmware.
I get a 'successful upload' through Arduino IDE,(with usb cable connected) both 1.8.x & 2.0.x ,,but the module never functions as it should regardless what sketch is uploaded to it.

Thank You

xidameng, 2021-07-29:
> FL0WL0W
> Am experiecing the same situation with two of the B&T rtlduino BW16 modules. The erasing procedure can not be done with a simple usb connected to the rtlduino,,,i am assuming? Am really [...]

@brcisna 

Hi this BW16 module has some strange setting to the flash that is preventing the MCU to boot from the image you just uploaded( they are already uploaded but placed in different memory segment as the default factory frimware which gets priority to boot every single time).

HOWEVER, this has been FIXED by our lovely community members with quite a few different methods, DO check out this link for details!

https://forum.amebaiot.com/t/resources-bw16-troubleshooting-guide/678

brcisna, 2021-07-29:
Thank You SimonXI,,
I will check out the link you provided..
I am old,,,and really struggle,,,trying to get my head around the Arduino
IDA stuff,,,and pinouts,, etc.

Thanks

xidameng, 2021-07-29:
> Thank You SimonXI,, I will check out the link you provided.. I am old,,,and really struggle,,,trying to get my head around the Arduino IDA stuff,,,and pinouts,, etc. Thanks
> […](#)

Sure thing, if you need pinout, you may check out this lovely post by the same community member who contributed the `troubleshooting guide` 

https://forum.amebaiot.com/t/resources-bw16-getting-start-guide/683

xidameng, 2021-07-29:
> Thank You SimonXI,, I will check out the link you provided.. I am old,,,and really struggle,,,trying to get my head around the Arduino IDA stuff,,,and pinouts,, etc. Thanks
> […](#)

Also, if you need a more timely reply and live discussion online, you may consider joinning us on Facebook @ https://www.facebook.com/groups/AmebaIoT

There are many Ameba engineers and maker on the FB group who will assist you with your questions faster

FL0WL0W, 2021-07-29:
More validation that the firmware uploader should be added under the tools drop down. Would make it easier for newcomers.

xidameng, 2021-07-29:
> More validation that the firmware uploader should be added under the tools drop down. Would make it easier for newcomers.

Please consider supporting this feature if you happen to have the skill/knowledge needed :)