Ameba-AIoT/ameba-arduino-d #34: Boot pins?

FL0WL0W, 2021-07-06:
please provide a table similar to below on what state the pins should be in on boot. I know pulling LOG_TX low will put it in UART Download Mode, but it appears pulling PA_27 low does something as well. I could not find any information in the datasheet about this either.

<pre>                                  \| GPIO 0 \| GPIO 2 \| GPIO 15 (MTDO)
--
  | ----------------------------------\|--------\|--------\|---------
  | Flash Startup (Normal)            \|   1    \|   1    \|   00
  | UART Download Mode (Programming)  \|   0    \|   1    \|   0
  | SD-Card Boot                      \|   0    \|   0    \|   1
  | </pre>

Aurical, 2021-07-08:
The UM0401 datasheet chapter 2.4 shows a list of pins that have special functions at boot. PA_27, in conjunction with PA_12 to PA_15, are used for Realtek's test mode, and do not seem to offer any functionality intended for an end-user

ambiot, 2021-07-16:
Proceed to close.

closed 2021-07-16 by ambiot