Ameba-AIoT/ameba-arduino-d #35: Add Software I2C support

FL0WL0W, 2021-07-06:
This is more of a feature request than an Issue. I know there is Hardware I2C which is fantastic! This is something that is lacking with the ESP8266 and I look forward to utilizing it in future projects. However, currently I am trying to reporpose some PCBs I had made for the ESP8266 to be used with the RTL8720DN (BW16) and it uses different pins. There is a SoftWire library that can be used, but I also have to update the subsequent sensor libraries to use SoftWire instead of the standard Wire interface. Please consider adding soft I2C support to Wire when selecting pins that do not have hardware support.

xidameng, 2021-07-07:
Hi Daniel. 

We are overhauling our I2C driver and might add in new features soon, stay tuned!

Meanwhile, I believe Software I2C is a pure software implementation, thus you may consider adding it to ameba yourself and we will see what we can do to assist you with your porting work 😃

FL0WL0W, 2021-07-07:
The SoftWire library works perfectly after fixing the pinMode error detailed in https://github.com/ambiot/ambd_arduino/issues/38

Leaving the Issue open as a feature request for softwire to be called directly from the I2C Wire library whenever hardware I2C is not available for the selected pins. This feature will make using other libraries in the arduino environment easier to use when using pins that do not have hardware I2C support.

closed 2021-07-16 by ambiot