Ameba-AIoT/ameba-arduino-d #37: SoftWire

FL0WL0W, 2021-07-06:
I am trying to run the ListDevices example from the SoftWire library that does I2C in software and bit bangs the digitalWrite/digitalRead to implement I2C. I know there is hardware I2C but my device is not connected to one of those pins. I am reusing a board made for an ESP8266 so i cannot change which pins it is connected to. I have defined pins 23 and 2 for SDA and SCL (PA_30 and PB_2). I do not see any toggling on the scope. I have verified that digitalWrite and digitalRead work for both pins. Maybe there is a limit to how fast the GPIO can be toggled?

xidameng, 2021-07-07:
Hi Daniel,

Thanks for your efforts, as of module specifications, you may refer to this document for more information on BW16,

https://bit.ly/RTL8720DN

If you need more peripheral specific information, you may check out our [online documentation](https://amebaiotdocuments.readthedocs.io/en/latest/ambd_arduino/ambd/download/index.html) to download `datasheet` as well as `applicaiton notes`, these should provide you with enough informaiton

FL0WL0W, 2021-07-07:
Hi xidameng, I created an account and verified my email. Looks like i do not have permission to download the information

FL0WL0W, 2021-07-07:
Softwire works perfectly after the fixes detailed in https://github.com/ambiot/ambd_arduino/issues/38

closed 2021-07-07 by FL0WL0W