Ameba-AIoT/ameba-arduino-d #38: g_APinDescription[ulPin].ulPinMode not updated after first call to pinMode(pin, mode)

FL0WL0W, 2021-07-07:
in wiring_digital.c line 93, 

```g_APinDescription[ulPin].ulPinMode = ulMode;```

this line is wrapped in the if NOT_INITIAL so this only gets updated on the first call to pinMode.

example

```
pinmode(PA30, OUTPUT); //g_APinDescription[ulPin].ulPinMode value updated. pin is configured
pinmode(PA30, INPUT); //g_APinDescription[ulPin].ulPinMode value NOT updated. pin is configured
pinmode(PA30, OUTPU); //g_APinDescription[ulPin].ulPinMode value NOT updated. pin is NOT configured, pinMode function is returned at line 56 in wiring_digital.c   pin is still setup as input.
```

moving line 93 to after line 96 in wiring_digital.c fixes the issue.

ambiot, 2021-07-09:
Dea this issue cause by the typo of "OUTPUT" in the third line?

FL0WL0W, 2021-07-09:
no it is not caused by the typo. i fixed the edit

FL0WL0W, 2021-07-09:
In wiring_digital.c

This
````
if (g_APinDescription[ulPin].ulPinType == NOT_INITIAL) {
        if (ulMode == INPUT_IRQ_FALL || ulMode == INPUT_IRQ_RISE || ulMode == INPUT_IRQ_LOW || ulMode == INPUT_IRQ_HIGH) {
            gpio_pin_struct[ulPin] = malloc (sizeof(gpio_irq_t));
            pGpio_t = gpio_pin_struct[ulPin];
            gpio_irq_init(pGpio_t, g_APinDescription[ulPin].pinname, gpioIrqHandler, ulPin);
            g_APinDescription[ulPin].ulPinType = PIO_GPIO_IRQ;
        } else {
            gpio_pin_struct[ulPin] = malloc (sizeof(gpio_t));
            pGpio_t = gpio_pin_struct[ulPin];
            gpio_init(pGpio_t, g_APinDescription[ulPin].pinname);
            g_APinDescription[ulPin].ulPinType = PIO_GPIO;
        }
        g_APinDescription[ulPin].ulPinMode = ulMode;	
    } else {
        pGpio_t = gpio_pin_struct[ulPin];
    }	
```

needs to be changed to this

```
if (g_APinDescription[ulPin].ulPinType == NOT_INITIAL) {
        if (ulMode == INPUT_IRQ_FALL || ulMode == INPUT_IRQ_RISE || ulMode == INPUT_IRQ_LOW || ulMode == INPUT_IRQ_HIGH) {
            gpio_pin_struct[ulPin] = malloc (sizeof(gpio_irq_t));
            pGpio_t = gpio_pin_struct[ulPin];
            gpio_irq_init(pGpio_t, g_APinDescription[ulPin].pinname, gpioIrqHandler, ulPin);
            g_APinDescription[ulPin].ulPinType = PIO_GPIO_IRQ;
        } else {
            gpio_pin_struct[ulPin] = malloc (sizeof(gpio_t));
            pGpio_t = gpio_pin_struct[ulPin];
            gpio_init(pGpio_t, g_APinDescription[ulPin].pinname);
            g_APinDescription[ulPin].ulPinType = PIO_GPIO;
        }
    } else {
        pGpio_t = gpio_pin_struct[ulPin];
    }	
    g_APinDescription[ulPin].ulPinMode = ulMode;	
```

xidameng, 2021-07-12:
Thanks Daniel, will test it on my setup and revert to u

ambiot, 2021-07-16:
Thank you! commit mentioned at #35  
Will merge into master branch next release. 
Processed to closed.

closed 2021-07-16 by ambiot