Ameba-AIoT/ameba-arduino-d #44: auto flash support

jojoling, 2021-08-05:
auto upload flash & reset system is a key feature for developer on ameba arduino.
https://github.com/jojoling/ameba_bw16_autoflash

amebad_image_tool issues:
1. It can work on windows 10.  But it has a issue that will get "error: Enter Uart Download Mode." only once when USB serial port device just plugin into computer.

2. Linux/macOS don't work due to "amebad_image_tool(no source)" also control RS232 signal DTR/RTS. It makes system leave uart download mode and get failed. It can see this situation on oscilloscope.

Thanks.

jojoling, 2021-08-05:
Below is windows 10 fail log in first time build and upload. I do changed binary offset 0x2B60 value from 0x2 to 0xF to see this.

>C:\Users\xxx\AppData\Local\Arduino15\packages\realtek\tools\ameba_d_tools\1.0.5/upload_image_tool_windows.exe C:\Users\xxx\AppData\Local\Arduino15\packages\realtek\tools\ameba_d_tools\1.0.5 COM8
Enter uart flash download mode..
Uploading.      setting serial port timeouts to 1000 ms
        flush start
        setting serial port timeouts to 1 ms
                flushed 0 bytes
        setting serial port timeouts to 1000 ms
        flush complete
        setting serial port timeouts to 1000 ms
        flush start
        setting serial port timeouts to 1 ms
                flushed 0 bytes
        setting serial port timeouts to 1000 ms
        flush complete
..      read 0, requested 1
        TIME SPENT=1391

..      TIME SPENT=2204

error: Enter Uart Download Mode
Image tool closed!
    Upload Image done.
Hard resetting via RTS pin..

jojoling, 2021-08-22:
Hi guys...

Does it has any updating for this issue?

xidameng, 2021-08-24:
Hi @jojoling 

Thanks for adding this handy feature, and Yes I believe it's a convenient feature and will potentially save your time and hassle, but this feature is indeed implemented via hardware which is not part of any Ameba EVB, it's heavy dependency on additional hardware makes it not suitable to be added to the SDK as majority of Ameba users would not have access/knowledge to add/troubleshoot this piece of hardware. 

In fact, in one of my local test, using FTDI USB to serial chip seems not able to work with the new implementation somehow...

Therefore, we will discuss this further when there is a need to design new Ameba EVB in the future and I believe your work will definitely be put into consideration :)

closed 2021-08-31 by M-ichae-l