Ameba-AIoT/ameba-arduino-d #54: Setting 10MHz SPI

ajsb85, 2021-09-09:
Dearest.

Can I get references of the SPI speed on Arduino?

Regards.

https://github.com/ambiot/ambd_arduino/blob/dfdc7767c2c880e344ae44cafe963478bfed4225/Arduino_package/hardware/libraries/SPI/SPI.cpp#L277-L283

xidameng, 2021-09-09:
Hi @ajsb85 

Check these 2 lines

https://github.com/ambiot/ambd_arduino/blob/dfdc7767c2c880e344ae44cafe963478bfed4225/Arduino_package/hardware/libraries/SPI/SPI.cpp#L46

https://github.com/ambiot/ambd_arduino/blob/dfdc7767c2c880e344ae44cafe963478bfed4225/Arduino_package/hardware/libraries/SPI/SPI.cpp#L114

ajsb85, 2021-09-09:
@xidameng thanks.

The default is 20MHz.
AmebaD support 50MHz and 25MHz.
Then for 10MHz:

SPI.beginTransaction(SPISettings(10000000, MSBFIRST, `SPI_MODE0))`

https://github.com/ambiot/ambd_arduino/blob/dfdc7767c2c880e344ae44cafe963478bfed4225/Arduino_package/hardware/libraries/SPI/SPI.cpp#L46

closed 2021-09-09 by ajsb85