Ameba-AIoT/ameba-arduino-d #59: When i setup on arduino，Not find ameba_d_tools_windows-1.0.6.tar.gz

xuhongv, 2021-11-06:
```
download https://github.com/ambiot/ambd_arduino/master/dev/Arduino_package/release/ameba_d_tools_windows-1.0.6.tar.gz  error
java.lang.RuntimeException: java.lang.Exception: download https://github.com/ambiot/ambd_arduino/master/dev/Arduino_package/release/ameba_d_tools_windows-1.0.6.tar.gz  error
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: download https://github.com/ambiot/ambd_arduino/master/dev/Arduino_package/release/ameba_d_tools_windows-1.0.6.tar.gz error
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:112)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:239)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more
```

ambiot, 2021-11-08:
Thanks for the feedback. 
The fix is updated. Please try the setup again.

xuhongv, 2021-11-08:
I am engineer from china ,  it is bad for me to setup on arduino , because the network in the github.
Please tell me how to  get the 3 files , how to Offline to setup：

```
If you are facing GitHub downloading issue, please refer to the following link at “Download/Software Development Kit”. There are 3 sections.
1. “AmebaD_Arduino_patch1_SDK”, please select at least 1 of the SDKs. There are 5 latest released SDK options.
2. “AmebaD_Arduino_patch2_Tools”, please select according to your operating system. There are Windows, Linux, and macOS.
3. “AmebaD_Arduino_Source_Code”, this section is an optional download only wants to refer to the latest source code.
```

@ambiot

xidameng, 2021-11-08:
> I am aithinker engineer from china ,the BW16 module is a product of my company. But it is bad for me to setup on arduino , because the network in the github. Please tell me how to get the 3 files , how to Offline to setup：
> 
> ```
> If you are facing GitHub downloading issue, please refer to the following link at “Download/Software Development Kit”. There are 3 sections.
> 1. “AmebaD_Arduino_patch1_SDK”, please select at least 1 of the SDKs. There are 5 latest released SDK options.
> 2. “AmebaD_Arduino_patch2_Tools”, please select according to your operating system. There are Windows, Linux, and macOS.
> 3. “AmebaD_Arduino_Source_Code”, this section is an optional download only wants to refer to the latest source code.
> ```
> 
> @ambiot

The link is here https://www.amebaiot.com.cn/cn/amebad-arduino-download/

closed 2021-11-09 by ambiot