Ameba-AIoT/ameba-arduino-d #62: Compile fails on Windows if space exists in path (eg username)

Norlin76, 2021-12-15:
fails on this copy command due to space in path

`cmd /c copy /y "C:\\Users\\Michael" "Gilchrist\\AppData\\Local\\Arduino15\\packages\\realtek\\tools\\ameba_d_tools\\1.0.6\\km0_km4_image2.bin" "C:\\Users\\MICHAE~1\\AppData\\Local\\Temp\\arduino_build_798621"
The system cannot find the file specified.
exit status 1
Error compiling for board RTL8720DN(BW16).
`

full output: [https://pastebin.com/4sm2K86K](url)

xidameng, 2021-12-18:
> Michael" "Gilchrist

Just to double check:
1.Are you using Arduino IDE to export compiled binary?
2. Is your PC's full user name `Michael Gilchrist`?

Norlin76, 2021-12-18:
1. Downloaded and installed arduino ide, went through steps to add board manager url, installed boards for realtek ameba, selected board RTL8720DN(BW16). Opened example 01.Basic -> Blink. Clicked check mark, got output attached on pastebin.
2. full username is "Michael Gilchrist"

xidameng, 2021-12-18:
> 1. Downloaded and installed arduino ide, went through steps to add board manager url, installed boards for realtek ameba, selected board RTL8720DN(BW16). Opened example 01.Basic -> Blink. Clicked check mark, got output attached on pastebin.
> 2. full username is "Michael Gilchrist"

roger that, for now, you can try this first,

1. Locate `Platform.txt` file inside your arduino ameba package, for your case, it should be at `C:\\Users\\Michael Gilchrist\\AppData\\Local\\Arduino15\\packages\\realtek\\hardware`
2. Scroll to this [line](https://github.com/ambiot/ambd_arduino/blob/b87e191c7350890c4bf45ce4198dac81959deb34/Arduino_package/hardware/platform.txt#L131) 
3. Change this line to this  `recipe.hooks.objcopy.postobjcopy.pattern.windows=cmd /c copy /y "{runtime.tools.ameba_d_tools.path}\km0_km4_image2.bin" {build.path}`
4. Compile a blink exmaple and check again

Norlin76, 2021-12-18:
Ahh, I was looking for some sort of batch file, thanks for pointing out which file I need to look at!

turns out I have platform.txt in 
C:\Users\Michael Gilchrist\AppData\Local\Arduino15\packages\realtek\tools\ameba_d_tools\1.0.6
and
C:\Users\Michael Gilchrist\Documents\ArduinoData\packages\realtek\hardware\AmebaD\3.1.0

had to edit the second location, interesting that it uses {runtime.tools.ameba_d_tools.path} many times but only messes up on this line. Regardless, adding the quotes fixed it, most excellent!

xidameng, 2021-12-19:
> Ahh, I was looking for some sort of batch file, thanks for pointing out which file I need to look at!
> 
> turns out I have platform.txt in C:\Users\Michael Gilchrist\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.0 and C:\Users\Michael Gilchrist\Documents\ArduinoData\packages\realtek\hardware\AmebaD\3.1.0
> 
> had to edit the second location, interesting that it uses {runtime.tools.ameba_d_tools.path} many times but only messes up on this line. Regardless, adding the quotes fixed it, most excellent!

yeah, I was also quite surprised that it only messed up that 1 single line not other places.

> adding the quotes fixed it

Anyway, glad it helps!

closed 2021-12-19 by xidameng