Ameba-AIoT/ameba-arduino-d #67: RTL8720D and Neopixel

ITstreet1, 2022-01-13:
I am pointed from the Ameba forum to open an issue here.

Here is the complete post:

---
I have RTL8720D and I tried to use some Neopixel with it. Neopixel is WS2812 and I connected it to pin 10. This is the sketch:

```
#include <Adafruit_NeoPixel.h>
#define PIN 10
Adafruit_NeoPixel strip = Adafruit_NeoPixel(1, PIN, NEO_GRB + NEO_KHZ800);
void setup() {
  strip.begin();
  strip.setBrightness(50);
  strip.show(); // Initialize all pixels to 'off'
}
void loop() {
  colorWipe(strip.Color(0, 255, 0), 1000); // Green
  //colorWipe(strip.Color(255, 0, 0), 1000); // Green
  //colorWipe(strip.Color(0, 0, 255), 1000); // Green
  colorWipe(strip.Color(0, 0, 0), 1000); // white
}

// Fill the dots one after the other with a color
void colorWipe(uint32_t c, uint8_t wait) {
  for(uint16_t i=0; i<strip.numPixels(); i++) {
      strip.setPixelColor(i, c);
      strip.show();
      delay(wait);
  }
}
```

With this setup, my Neopixel does not work. What am I doing wrong? I use Arduino IDE. I can blink the regular LED on that pin. What could be wrong with Neopixel? Should I use some other pin?
---
So, I am told that SDK is not supporting WS2812. Is there any chance to support it in near future?

S10143806H, 2022-01-27:
Hi @ITstreet1,
Here is the example working on AMB01 AMB02 https://www.amebaiot.com/en/ws2812b-4x4-rgb-led/.
We are currently trying to port the feature to Ameba D (AMB21/22/23). 
We will update you again once there is any update. 

Thank you.

ITstreet1, 2022-01-27:
Thank you in advance.

daphwl, 2022-02-25:
Hi @ITstreet1, 
 
Sorry to tell you but after some testing and confirming with the development team, because of the hardware limitation, the toggling speed of the AmebaD's GPIO will not be fast enough to drive the NeoPixel LED. So it can't be port to AmebaD. If you want to use the NeoPixel, you may consider using the working example on AMB01 AMB02. 

Thank you.

ITstreet1, 2022-02-25:
If I get you correctly, I can drive Neopixel with RTL00, which is RTL8710?

daphwl, 2022-02-25:
Yes can

daphwl, 2022-02-25:
But not on AmebaD[RTL8722DM/RTL8722CSM/RTL8722DM MINI/ BW16].

ITstreet1, 2022-02-25:
RTL8711, RTL8195, and RTL8710 are ok?

daphwl, 2022-02-25:
Yes

xidameng, 2022-02-28:
> RTL8711, RTL8195, and RTL8710 are ok?

you are, you can see from this video that uses RTL8710 to drive the neopixel LED(WS2812) flawlessly 
https://www.facebook.com/100002223592575/videos/664430788141270/

ITstreet1, 2022-02-28:
What did you use to program this board? Arduino?

ambiot, 2022-02-28:
Yes, Arduino SDK. https://github.com/ambiot/amb1_arduino

ambiot, 2022-03-14:
Currently, there is another drive under developing for this hardware, but it is developed by using SPI. The speed of SPI is faster enough for driving it.

ambiot, 2022-05-10:
Please refer to #76 for the new feature supported.

closed 2022-05-10 by ambiot