Ameba-AIoT/ameba-arduino-d #73: Dependency libraries

Timothychen00, 2022-02-08:
1.I was using esp series like esp8266 to make some Iot server project. And so that i have the use of Async Servers whitch is a library of esp8266 but when i turn to rtl8720dn for the support of 5G wifi, there is no such lib for rtl8720dn(because the libs of esp8266 use some base APIs that rtl8720dn doesnt have) to more efficiently handle those net connections.  

2.Are you going to support file systems like SPIFFS or LittleFS in esp8266. because it would be a good choice to have a easier to used file systems to store file or data for local storage.(because some users isn't able to have a internet service to process the datas whitch came from the IOT devices.)?

S10143806H, 2022-09-05:
Hi @Timothychen00 ,
1. For Async Servers, currently Ameba Arduino does not support handling more than one connection at the same time. However, you may check [WiFiManager_RTL8720](https://github.com/khoih-prog/WiFiManager_RTL8720) for more information since his porting allows Ameba to connect to more than one AP.
2. For file system support, currently we only support [FATFS](https://github.com/ambiot/ambd_arduino/tree/dev/Arduino_package/hardware/libraries/FatfsSDIO). 
Hope answers here would help.
Thanks.

xidameng, 2022-10-17:
Closing this issue as there is no plan to add supports for the requested features

closed 2022-10-17 by xidameng

bkrajendra, 2022-11-13:
It will be good to have Flash File storage (SPIFFS)
SD card is always not a good choice for small and less frequent data-like configurations.

vshymanskyy, 2023-08-29:
For configuration storage, see https://github.com/ambiot/ambd_arduino/issues/195#issuecomment-1669381942