Ameba-AIoT/ameba-arduino-d #75: BW16 MQTT disconnect issue

Brian9871, 2022-02-23:
Hi,
Has this SDK been tested MQTT for a long time ?
After testing, it is found that turning on Gtimer and regularly performing MQTT-publish will cause irregular disconnection.
This problem occurs even though Gtimer is turned off, but not as frequently.
Hope this part can be improved.

Brian9871, 2022-03-03:
Hi,
I found a reason for the disconnection,
When using UART and MQTT in the program at the same time, it will lead to irregular disconnection.
I hope there will be a solution to make it work.
Thank you.

[image]

daphwl, 2022-03-03:
Hi @Brian9871,

I have tried to test the MQTT and I don't see any disconnection issue. So I'm not sure if there's an issue with your code. Thank you.

Brian9871, 2022-03-03:
Hi @daphwl ,
The disconnection happens in about 5-10 minutes, when my UART transmits at 5hz.

daphwl, 2022-03-03:
Are you using the Arduino SDK MQTT example?

Brian9871, 2022-03-03:
Yes, I use the Arduino SDK MQTT example, and add serial function to it.
there will be a problem of disconnection, I attach my code here.

```
/*
 Basic MQTT example

 This sketch demonstrates the basic capabilities of the library.
 It connects to an MQTT server then:
  - publishes "hello world" to the topic "outTopic"
  - subscribes to the topic "inTopic", printing out any messages
    it receives. NB - it assumes the received payloads are strings not binary

 It will reconnect to the server if the connection is lost using a blocking
 reconnect function. See the 'mqtt_reconnect_nonblocking' example for how to
 achieve the same result without blocking the main loop.
 
*/

#include <WiFi.h>
#include <PubSubClient.h>
#include <SoftwareSerial.h>

// Update these with values suitable for your network.

char ssid[] = "CJS_5G";     // your network SSID (name)
char pass[] = "77777777";  // your network password
int status  = WL_IDLE_STATUS;    // the Wifi radio's status

char mqttServer[]     = "192.168.137.1";
char clientId[]       = "amebaClient";
char publishTopic[]   = "outTopic";
char publishPayload[] = "hello world";
char subscribeTopic[] = "inTopic";

/* UART */
SoftwareSerial mySerial(PB2, PB1); // RX, TX

void callback(char* topic, byte* payload, unsigned int length) {
    mySerial.print("Message arrived [");
    mySerial.print(topic);
    mySerial.print("] ");
    for (int i = 0; i < length; i++) {
        mySerial.print((char)(payload[i]));
    }
    mySerial.println();
}

WiFiClient wifiClient;
PubSubClient client(wifiClient);

void reconnect() {
    // Loop until we're reconnected
    while (!(client.connected())) {
        mySerial.print("Attempting MQTT connection...");
        // Attempt to connect
        if (client.connect(clientId)) {
            mySerial.println("connected");
        } else {
            mySerial.print("failed, rc=");
            mySerial.print(client.state());
            mySerial.println(" try again in 5 seconds");
            // Wait 5 seconds before retrying
            delay(5000);
        }
    }
}

void setup() {
    mySerial.begin(1000000);

    while (status != WL_CONNECTED) {
        mySerial.print("Attempting to connect to SSID: ");
        mySerial.println(ssid);
        // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
        status = WiFi.begin(ssid, pass);

        // wait 10 seconds for connection:
        delay(10000);
    }

    client.setServer(mqttServer, 1883);
    client.setCallback(callback);

    // Allow the hardware to sort itself out
    delay(1500);
}
long nowTime = 0;
long lastTime = 0;
void loop() {
    nowTime = millis();
    if (nowTime > (lastTime + 200))
    {
      mySerial.write(0x84);
      client.publish(publishTopic, publishPayload);
      lastTime = nowTime;
    }
    client.loop();
    if (!(client.connected())) {
        reconnect();
    }
}
```

daphwl, 2022-03-03:
Hi, 
I have taken a look at your code. Saw that you have changed all the ` Serial` into ``mySerial`. I would recommend that you test out the default MQTT example code that we have in Arduino SDK first and double-check if it is working because I have tested that MQTT example and it working on my side. Thank you.

Brian9871, 2022-03-03:
Hi,
I have tried the Serial in this program, and still have the problem.
and it happened in 5-15min.
Can you run this code for at least twenty minutes and let me know if there is any disconnection?
Thank you very much.

```
/*
 Basic MQTT example

 This sketch demonstrates the basic capabilities of the library.
 It connects to an MQTT server then:
  - publishes "hello world" to the topic "outTopic"
  - subscribes to the topic "inTopic", printing out any messages
    it receives. NB - it assumes the received payloads are strings not binary

 It will reconnect to the server if the connection is lost using a blocking
 reconnect function. See the 'mqtt_reconnect_nonblocking' example for how to
 achieve the same result without blocking the main loop.
 
*/

#include <WiFi.h>
#include <PubSubClient.h>

// Update these with values suitable for your network.

char ssid[] = "CJS_5G";     // your network SSID (name)
char pass[] = "77777777";  // your network password
int status  = WL_IDLE_STATUS;    // the Wifi radio's status

char mqttServer[]     = "192.168.137.1";
char clientId[]       = "amebaClient";
char publishTopic[]   = "outTopic";
char publishPayload[] = "hello world";
char subscribeTopic[] = "inTopic";

void callback(char* topic, byte* payload, unsigned int length) {
    Serial.print("Message arrived [");
    Serial.print(topic);
    Serial.print("] ");
    for (int i = 0; i < length; i++) {
        Serial.print((char)(payload[i]));
    }
    Serial.println();
}

WiFiClient wifiClient;
PubSubClient client(wifiClient);

void reconnect() {
    // Loop until we're reconnected
    while (!(client.connected())) {
        Serial.print("Attempting MQTT connection...");
        // Attempt to connect
        if (client.connect(clientId)) {
            Serial.println("connected");
        } else {
            Serial.print("failed, rc=");
            Serial.print(client.state());
            Serial.println(" try again in 5 seconds");
            // Wait 5 seconds before retrying
            delay(5000);
        }
    }
}

void setup() {
    Serial.begin(115200);

    while (status != WL_CONNECTED) {
        Serial.print("Attempting to connect to SSID: ");
        Serial.println(ssid);
        // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
        status = WiFi.begin(ssid, pass);

        // wait 10 seconds for connection:
        delay(10000);
    }

    client.setServer(mqttServer, 1883);
    client.setCallback(callback);

    // Allow the hardware to sort itself out
    delay(1500);
}
long nowTime = 0;
long lastTime = 0;
void loop() {
    nowTime = millis();
    if (nowTime > (lastTime + 200))
    {
      Serial.write(0x84);
      client.publish(publishTopic, publishPayload);
      lastTime = nowTime;
    }
    client.loop();
    if (!(client.connected())) {
        reconnect();
    }
}
```

ambiot, 2022-03-03:
@Brian9871 
Hi, are you able to connect the **default** example? **Arduino_package/hardware/libraries/MQTTClient/examples/MQTT_Basic/MQTT_Basic.ino**

Brian9871, 2022-03-03:
Hi, @ambiot ,
But in my project, I need to use TX_0 and RX_0 to received sensor data.
Is it possible to achieve the myserial program I provided above?

Brian9871, 2022-03-03:
@ambiot The TX_0 and RX_0 means that Serial_TX and Serial_RX.
If you have the BW16 kit, you can see TX_0 and RX_0 written on it...

All in all, I hope to use Serial RX/TX to receive data and upload it through MQTT. 
However, at present, when SoftSerial and MQTT are used at the same time, there will be occasional disconnections.
The code just uses example and SoftSerial to cause problems. I hope someone can provide relevant assistance.
Thanks.

Brian9871, 2022-03-04:
https://github.com/ambiot/ambd_arduino/issues/75#issuecomment-1057773950
I hope someone can confirm first to see if there is any problem with this code, but this only integrates the MQTT Example and the SoftSerial Example.
This code structure is also very simple, but there will be a disconnection problem in 5-20 minutes.

Brian9871, 2022-03-08:
https://github.com/ambiot/ambd_arduino/issues/75#issuecomment-1057773950
Can anyone reproduce my problem? I've tried both BW16 kits with the same problem.
Hopefully someone can try my code and discuss where the problem might be.
I have done many functions on BW16 such as Webserver, OTA, etc.
But if this problem can't be solved, I can only give up using this module...

daphwl, 2022-07-27:
Issue fixed with #101

closed 2022-07-27 by Aurical

autonomouscitizen, 2024-09-02:
> Issue fixed with #101

this is definitely not fixed. still experiencing this issue in 2024