Ameba-AIoT/ameba-arduino-d #81: I2C Slave API fails to recognize the master's final NACK in slave-to-master transmissions

xavier862, 2022-05-17:
This is a replicate of [this issue](https://github.com/ambiot/ambd_sdk/issues/8), which was corrected in the AmebaD SDK but unfortunately left the Arduino SDK aside. [The buggy I2C_SlaveWrite()](https://github.com/ambiot/ambd_arduino/blob/4823c9d7aa80c4bcc755857e1984a50c11ddfc3a/Arduino_package/hardware/system/component/soc/realtek/amebad/fwlib/ram_common/rtl8721d_i2c.c#L719) in the Arduino SDK should be replaced with [this corrected version](https://github.com/ambiot/ambd_sdk/blob/76daaaaa550c02cbfdd8a11e5023d89b235fd9d8/component/soc/realtek/amebad/fwlib/ram_common/rtl8721d_i2c.c#L719) of the AmebaD SDK.

xidameng, 2022-08-03:
Hi @xavier862 , 

Thanks for your feedback and kind help to point out the issue, the good news is,

The `I2C_SlaveWrite()` has been corrected in the early release version, you may preview the changes [here ](https://github.com/ambiot/ambd_arduino/blob/7648a72aee6bc4a90820e5cf559b41f15ae66b62/Arduino_package/hardware/system/component/soc/realtek/amebad/fwlib/ram_common/rtl8721d_i2c.c#L733)and feature updates using the `early release` URL at,
https://github.com/ambiot/ambd_arduino/blob/dev/Arduino_package/package_realtek.com_amebad_early_index.json

---

In addition to that, I2C slave mode (both slave reader and sender) has been added to the Arduino SDK under dev branch( #105  ) , it will be officially added to Arduino package in the upcoming release

closed 2022-08-03 by xidameng

xavier862, 2022-08-05:
Hi Simon,

This is good news ! I will have a look at the Arduino I2C slave mode.
Cheers