Ameba-AIoT/ameba-arduino-d #86: The application does not end normally.

Oliver0804, 2022-06-06:
There are always ameba_image_tools running even after the Arduino is closed.

Arduino 1.8.3
macOS  12.3.1

[image]

M-ichae-l, 2022-06-15:
Hi @Oliver0804 . 
I have tested it on the macOS. I did not get this issue. You can double check you device after update all your software. 

- Arduino 1.8.19
- macOS 12.4
- Activity Monitor

ambiot, 2022-07-01:
Hi @Oliver0804 , 
Dose the issue still appear with V3.1.3 SDK? 
@M-ichae-l is not able to reproduce the issue, let us double check with the new version of SDK.

Oliver0804, 2022-07-01:
I tested version 3.1.3 and it's good.

closed 2022-07-01 by Oliver0804