Ameba-AIoT/ameba-arduino-d #87: WiFi.status() not updating bug

khoih-prog, 2022-06-07:
The current AmebaD core v3.1.2 has a bug that once WiFi is connected, `WiFi.status()` always reports **WL_CONNECTED** even if WiFi is lost. `WiFi.RSSI()` is also still keeping the good value.

Check the discussion in [**Solve WiFi.status() not updating bug** #1](https://github.com/khoih-prog/WiFiManager_RTL8720/pull/1) for more information.

S10143806H, 2022-06-08:
Hi @khoih-prog,

This bug has been fixed via updating `cores/ambd//main.cpp`: https://github.com/ambiot/ambd_arduino/blob/989e6e83d2820c43dadf9885700a1de482db0147/Arduino_package/hardware/cores/ambd/main.cpp#L117

Please assist with further verification from your end. 

I will proceed to close the [Pull Request](https://github.com/khoih-prog/WiFiManager_RTL8720/pull/1) in the original repo [WiFiManager_RTL8720](https://github.com/khoih-prog/WiFiManager_RTL8720) since it's only a walk-around fix.

Thanks.

khoih-prog, 2022-06-17:
Hi @S10143806H 

The bug is not yet fixed by changing only the priority 

https://github.com/ambiot/ambd_arduino/blob/989e6e83d2820c43dadf9885700a1de482db0147/Arduino_package/hardware/cores/ambd/main.cpp#L117

I have to add a vTaskDelay(1) tempo fix after

https://github.com/ambiot/ambd_arduino/blob/989e6e83d2820c43dadf9885700a1de482db0147/Arduino_package/hardware/cores/ambd/main.cpp#L93

```
    for (;;) {
        loop();
        if (serialEventRun) {
            serialEventRun();
        }
        osThreadYield();
        //vTaskDelete(NULL);
        vTaskDelay(1);
    }
```

so that the WiFi.status() can work normally

---

The test code is as follows

#### 1. WiFiStatusTest.ino

```
#include "defines.h"

int status = WL_IDLE_STATUS;     // the Wifi radio's status

void printWifiStatus()
{
  // print the SSID of the network you're attached to:
  // you're connected now, so print out the data
  Serial.print(F("You're connected to the network, IP = "));
  Serial.println(WiFi.localIP());

  Serial.print(F("SSID: "));
  Serial.print(WiFi.SSID());

  // print the received signal strength:
  int32_t rssi = WiFi.RSSI();
  Serial.print(F(", Signal strength (RSSI):"));
  Serial.print(rssi);
  Serial.println(F(" dBm"));
}

void heartBeatPrint()
{
  static int num = 1;

  if (WiFi.status() == WL_CONNECTED)
  {
    Serial.print("H");        // H means connected to WiFi
    // Bad bug or RTL8720 => if WiFi lost, WiFi.status() still is WL_CONNECTED and WiFi.RSSI() still OK.
    // Similar bug in Portenta_H7 without workaround as in Portenta_H7, WiFi.RSSI() => 0
    Serial.print(", RSSI ="); Serial.println(WiFi.RSSI());
  }
  else
    Serial.print("F");        // F means not connected to WiFi
}

void check_status()
{
  static unsigned long checkstatus_timeout = 0;

  //KH
#define HEARTBEAT_INTERVAL    10000L
  // Print hearbeat every HEARTBEAT_INTERVAL (10) seconds.
  if ((millis() > checkstatus_timeout) || (checkstatus_timeout == 0))
  {
    heartBeatPrint();
    checkstatus_timeout = millis() + HEARTBEAT_INTERVAL;
  }
}

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial.print(F("\nStarting WiFiStatusTest on "));  Serial.print(BOARD_NAME);
  Serial.print(F(" with ")); Serial.println(SHIELD_TYPE);
  Serial.println(WIFI_WEBSERVER_RTL8720_VERSION);

  if (WiFi.status() == WL_NO_SHIELD)
  {
    Serial.println(F("WiFi shield not present"));
    // don't continue
    while (true);
  }

  String fv = WiFi.firmwareVersion();

  Serial.print("Current Firmware Version = "); Serial.println(fv);

  if (fv != LATEST_RTL8720_FIRMWARE)
  {
    Serial.println("Please upgrade the firmware");
  }

  // attempt to connect to Wifi network:
  while (status != WL_CONNECTED)
  {
    Serial.print("Attempting to connect to SSID: "); Serial.println(ssid);

    // Connect to WPA/WPA2 network. 2.4G and 5G are all OK
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }

  // you're connected now, so print out the data
  printWifiStatus();
}

void loop()
{
  check_status();
}
```

#### 2. defines.h

```
#ifndef defines_h
#define defines_h

#if !defined(CONFIG_PLATFORM_8721D)
  #error Only for Ameba Realtek RTL8720DN, RTL8722DM and RTL8722CSM platform.
#endif

#define DEBUG_WIFI_WEBSERVER_PORT   Serial

// Debug Level from 0 to 4
#define _WIFI_LOGLEVEL_             3

#define BOARD_TYPE      "Rtlduino RTL8720DN"

#ifndef BOARD_NAME
  #if defined(ARDUINO_BOARD)
    #define BOARD_NAME    ARDUINO_BOARD
  #elif defined(BOARD_TYPE)
    #define BOARD_NAME    BOARD_TYPE
  #else
    #define BOARD_NAME    "Unknown Board"
  #endif  
#endif

#define SHIELD_TYPE       "RTL8720DN"

#include <WiFiWebServer_RTL8720.h>

char ssid[] = "YOUR_SSID";        // your network SSID (name)
char pass[] = "12345678";        // your network password

#endif    //defines_h
```

and the result, when unplugging the WiFi AP is

```
Starting WiFiStatusTest on Rtlduino RTL8720DN with RTL8720DN
WiFiWebServer_RTL8720 v1.1.2
interface 0 is initialized

interface 1 is initialized

Initializing WIFI ...

WIFI initialized

Current Firmware Version = 1.0.0
Attempting to connect to SSID: HueNet1

RTL8721D[Driver]: set ssid [HueNet1] 

RTL8721D[Driver]: rtw_set_wpa_ie[1160]: AuthKeyMgmt = 0x2 

RTL8721D[Driver]: rtw_restruct_sec_ie[4225]: no pmksa cached 

Interface 0 IP address : 192.168.2.132You're connected to the network, IP = 192.168.2.132
SSID: HueNet1, Signal strength (RSSI):-37 dBm
H, RSSI =-37
H, RSSI =-37
H, RSSI =-36
H, RSSI =-36
H, RSSI =-36 <============= Power off the WiFi AP to simulate the WiFi lost

RTL8721D[Driver]: no beacon for a long time, disconnect or roaming
F
```

Without the vTaskDelay(1), the WiFi.status is always OK even WiFi AP is powered off.

I believe **only Realtek can find out and have the final fix** for this as we don't have access to all the code to fix.

S10143806H, 2022-06-21:
Hi @khoih-prog ,

Could you please modify the following line in `AmebaD\3.1.2\cores\arduino\main.cpp` and test whether this would help with solving this issue?

```
osThreadDef(main_task, osPriorityLow, 1, MAIN_THREAD_STACK_SIZE);
```

Without further adding the `vTaskDelay(1)`, dropping down the `osPriority` to **`osPriorityLow`** will solve the issue. 

### Testing code

`WiFiStatusTest.ino` as provided above.

### Log
```
#calibration_ok:[2:19:11] 
15:34:15.716 -> 
15:34:15.716 -> Starting WiFiStatusTest on Rtlduino RTL8720DN with RTL8720DN
15:34:15.716 -> WiFiWebServer_RTL8720 v1.1.2
15:34:15.716 -> interface 0 is initialized
15:34:15.716 -> 
interface 1 is initialized
15:34:15.716 -> 
15:34:15.716 -> 

Initializing WIFI ...
15:34:15.855 -> 

WIFI initialized
15:34:15.855 -> 
Current Firmware Version = 1.0.0
15:34:15.855 -> Attempting to connect to SSID: QQZ
15:34:15.855 -> 
15:34:15.855 -> 

RTL8721D[Driver]: set ssid [QQZ] 
15:34:15.855 -> 
Attempting to connect to SSID: QQZ
15:34:30.068 -> 
15:34:30.068 -> 

RTL8721D[Driver]: set ssid [QQZ] 
15:34:30.068 -> 
Attempting to connect to SSID: QQZ
15:34:44.257 -> 
15:34:44.257 -> 

RTL8721D[Driver]: set ssid [QQZ] 
15:34:44.257 -> 
Attempting to connect to SSID: QQZ
15:34:58.467 -> 
15:34:58.467 -> 

RTL8721D[Driver]: set ssid [QQZ] 
15:34:58.467 -> 
Attempting to connect to SSID: QQZ
15:35:12.707 -> 
15:35:12.707 -> 

RTL8721D[Driver]: set ssid [QQZ] 
15:35:12.707 -> 
15:35:16.785 -> 

RTL8721D[Driver]: rtw_set_wpa_ie[1160]: AuthKeyMgmt = 0x2 
15:35:16.785 -> 
15:35:17.377 -> 

RTL8721D[Driver]: rtw_restruct_sec_ie[4225]: no pmksa cached 
15:35:17.425 -> 
15:35:17.556 -> 

RTL8721D[Driver]: start auth to 82:a7:c6:d9:34:d3
15:35:17.594 -> 
15:35:17.594 -> 

RTL8721D[Driver]: auth alg = 2
15:35:17.594 -> 
15:35:17.594 -> 

RTL8721D[Driver]: 
15:35:17.594 -> 
OnAuthClient:algthm = 0, seq = 2, status = 0, sae_msg_len = 0
15:35:17.594 -> 
15:35:17.594 -> 

RTL8721D[Driver]: auth success, start assoc
15:35:17.594 -> 
15:35:17.594 -> 

RTL8721D[Driver]: association success(res=9)
15:35:17.594 -> 
15:35:17.641 -> 

RTL8721D[Driver]: ClientSendEAPOL[1624]: no use cache pmksa 
15:35:17.641 -> 
15:35:17.641 -> 

RTL8721D[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)
15:35:17.641 -> 
15:35:17.641 -> 

RTL8721D[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
15:35:17.641 -> 
15:35:18.239 -> 

Interface 0 IP address : 192.168.43.243You're connected to the network, IP = 192.168.43.243
15:35:28.271 -> SSID: QQZ, Signal strength (RSSI):-36 dBm
15:35:28.271 -> H, RSSI =-36
15:35:38.246 -> H, RSSI =-38
15:35:48.298 -> H, RSSI =-39
15:35:58.277 -> H, RSSI =-38
15:36:08.262 -> H, RSSI =-38
15:36:18.315 -> H, RSSI =-38
15:36:21.722 ->                                     <============= Power off the WiFi AP to simulate the WiFi lost
15:36:21.722 -> 

RTL8721D[Driver]: no beacon for a long time, disconnect or roaming
15:36:21.722 -> 
FFFFFFFFFFFFFFFFFFFFFFFFFFFF
```

S10143806H, 2022-06-22:
Hi @khoih-prog ,

I do understand your concerns. Before answering your questions, I would like to clarify that the `osPriority` I have indicated in my comment above requires to be converted from CMSIS type osPriority to FreeRTOS priority number using the function below:  https://github.com/ambiot/ambd_arduino/blob/5c9b3435cd21704157ceb719780f97e6c02b8c2b/Arduino_package/hardware/system/component/os/freertos/cmsis_os.c#L40
with `fpriority += (priority - osPriorityIdle);`. 

Therefore, the current main task priority is set to **"1" NOT "-2"**.

------

1. WiFi related functions were developed by a separate team, and the priority is set to be quite low.
2. There's nothing called "default priority" 
3. You can refer to osPriority set in `main.cpp` of  [ESP32](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp)/[ESP8266](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_main.cpp#L43) both has been set to "1"
4. The reason why task switching to lower priority task (such as WiFi) is due to `osThreadYield();` but not `vTaskDelay()` is being used in the `main_taks()`:    https://github.com/ambiot/ambd_arduino/blob/380b479fe7aaa4cd0f05b0bfbb8bc3ca71ce3fb4/Arduino_package/hardware/cores/ambd/main.cpp#L93

  - `osThreadYield`/`taskYIELD` only will pass control to the next thread that is in the READY state.
  - `vTaskDelay` is a longer function that calculates a wake time, and blocks the task. It contains `vTaskSuspendAll()`  that suspends the scheduler: 
   [image]

Thank you.

khoih-prog, 2022-06-22:
Hi @S10143806H 

> WiFi related functions were developed by a separate team, and the priority is set to be quite low.

This explains the real issue. Inefficient `vTaskDelay` is sure not my solution, just to show that there is something wrong with the core priority as I really don't know and expect the WiFi priority is so low, not even `osPriorityNormal` or `osPriorityBelowNormal`

If you're sure now all is OK with the **priority**, so use it as the final solution. 

More tests are always better. I'll let you to close the issue when comfortable that all is OK.

PS:

It's better to have a note somewhere about the priorities of tasks (maintask, **WiFi**, etc.) because as a habit, without any warning, users can create certain new tasks, and use `osPriorityNormal` (///< priority: normal (default)) and can create unexpected issues.

S10143806H, 2022-06-23:
Hi @khoih-prog,

Upon some internal tests on different peripherals, changing the priority might not affect other examples. 

Besides, I have sent another PR to update the priority in the `dev` branch:
https://github.com/ambiot/ambd_arduino/blob/ce9829875141813c78a4ecb9ec3f3cc35af148d2/Arduino_package/hardware/cores/ambd/main.cpp#L117

You may proceed to close this issue.

> By the way, you may join our [AmebaIoT forum](https://forum.amebaiot.com/) or AmebaIoT FB [ZH](https://www.facebook.com/groups/AmebaIoT)/[EN](https://www.facebook.com/groups/amebaioten) to get the latest update of Ameba related news or raise any questions/bugs.

Thank you for your support. Keep in touch!

closed 2022-06-23 by khoih-prog