Ameba-AIoT/ameba-arduino-d #114: EAGAIN used wrongly in WiFiClient.cpp available() function

S10143806H, 2022-08-24:
This issue was found and proposed by Ameba Arduino SDK user @jojoling regarding the PR in https://github.com/ambiot/ambd_arduino/pull/101/files. 

@jojoling has mentioned the using of EAGAIN is incorrect in this modification, no function is handling EAGAIN (if there is no data available right now, try again later) according to the definition.

ambiot, 2022-08-25:
@jojoling can you give more details?

jojoling, 2022-08-25:
EAGAIN it mean "error, try again".
It may have no problem in available function. but it cause time delay.

EAGAIN it also happen in other situation. Should it be ignore and wait for next ?

S10143806H, 2022-08-26:
Hi @jojoling ,
Modification according to your issue feedback was updated in this PR: https://github.com/ambiot/ambd_arduino/pull/112/commits/8ac392a9aca9f2c3aee83cfcdf40d165b53b8953
Please check. 
Thank you.

closed 2022-09-02 by S10143806H