Ameba-AIoT/ameba-arduino-d #131: RTL8720DN did not work with Arduino MPU6050 library

rkuo2000, 2022-10-06:
MPU6050 & MPU9250 are very popular IMUs (3-axis accelerator, 3-axis gyroscope, 3-axis compass)
The [Arduino MPU6050 library](https://www.arduino.cc/reference/en/libraries/mpu6050/) works well with ESP32, but hang if using RTL8720DN.

[https://github.com/ElectronicCats/mpu6050](https://github.com/ElectronicCats/mpu6050)

Could you tell why this [I2Cdev.cpp](https://github.com/ElectronicCats/mpu6050/blob/master/src/I2Cdev.cpp) are not compatible with RTL8720DN ? and find a fix to make RTL8720DN working with MPU6050 / MPU9250.

xidameng, 2022-10-18:
Hi @rkuo2000 

Thanks for reaching out to us. May I ask if you can elaborate on how it "hang is using RTL8720DN"?

Because, after fixing a small `compilation issue` (this will be fixed soon), the library you mentioned works well with RTL8720DN, no sign of hanging. Just FYI, I have tried both `MPU6050_raw` and `IMU_Zera` examples, both works fine.

If you don't mind, you can also try `Adafruit_MPU6050` library which also works well with RTL8720DN.

rkuo2000, 2022-10-18:
Board: RTL8720DN (BW16-typeC)
IMU Library : MPU6050 0.6.0 by Electronic Cat (this library supports DMP)
IMU Code :  [MPU6050_DMP6_Teapot.ino](https://github.com/rkuo2000/Arduino/blob/master/examples/IMU/MPU6050_DMP6_Teapot/MPU6050_DMP6_Teapot.ino)  (copied from example in MPU6050 library

The hang situation maybe it is because of using MPU6050 v0.5.0 arduino library,
after I install v0.6.0,

1. Compilation error: 'digitalPinToInterrupt' was not declared in this scope

After commented the interrupt pin calls, I got

2. /home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3379:2: note: in In file included from /home/rkuo/Documents/Arduino/examples/IMU/MPU6050_DMP6_Teapot/MPU6050_DMP6_Teapot.ino:14:0:
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050_6Axis_MotionApps_V6_12.h: In member function 'uint8_t MPU6050::dmpProcessFIFOPacket(const unsigned char*)':
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050_6Axis_MotionApps_V6_12.h:578:60: warning: unused parameter 'dmpData' [-Wunused-parameter]
 uint8_t MPU6050::dmpProcessFIFOPacket(const unsigned char *dmpData) {
                                                            ^~~~~~~
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp: In member function 'void MPU6050::getMotion9(int16_t*, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*)':
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:1778:113: warning: unused parameter 'mx' [-Wunused-parameter]
 void MPU6050::getMotion9(int16_t* ax, int16_t* ay, int16_t* az, int16_t* gx, int16_t* gy, int16_t* gz, int16_t* mx, int16_t* my, int16_t* mz) {
                                                                                                                 ^~
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:1778:126: warning: unused parameter 'my' [-Wunused-parameter]
 void MPU6050::getMotion9(int16_t* ax, int16_t* ay, int16_t* az, int16_t* gx, int16_t* gy, int16_t* gz, int16_t* mx, int16_t* my, int16_t* mz) {
                                                                                                                              ^~
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:1778:139: warning: unused parameter 'mz' [-Wunused-parameter]
 void MPU6050::getMotion9(int16_t* ax, int16_t* ay, int16_t* az, int16_t* gx, int16_t* gy, int16_t* gz, int16_t* mx, int16_t* my, int16_t* mz) {
                                                                                                                                           ^~
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp: In member function 'void MPU6050::PrintActiveOffsets()':
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3358:197: error: 'dtostrf' was not declared in this scope
 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt
                                                                                                                                                                                                     ^
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3372:2: note: in expansion of macro 'printfloatx'
  printfloatx("", Data[0], 5, 0, ",  ");
  ^~~~~~~~~~~
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3358:197: error: 'dtostrf' was not declared in this scope
 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt
                                                                                                                                                                                                     ^
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3373:2: note: in expansion of macro 'printfloatx'
  printfloatx("", Data[1], 5, 0, ",  ");
  ^~~~~~~~~~~
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3358:197: error: 'dtostrf' was not declared in this scope
 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt
                                                                                                                                                                                                     ^
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3374:2: note: in expansion of macro 'printfloatx'
  printfloatx("", Data[2], 5, 0, ",  ");
  ^~~~~~~~~~~
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3358:197: error: 'dtostrf' was not declared in this scope
 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt
[... cut ...]
 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt
                                                                                                                                                                                                     ^
/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3379:2: note: in expansion of macro 'printfloatx'
  printfloatx("", Data[2], 5, 0, "\n");
  ^~~~~~~~~~~

exit status 1

Compilation error: exit status 1

rkuo2000, 2022-10-18:
Board: RTL8720DN (BW16-TypeC)
IDE: Arduino 2.0.0
IMU library : MPU6050 v0.6.0
Sample Code : [MPU6050_DMP6_Teapot.ino](
https://github.com/rkuo2000/Arduino/tree/master/examples/IMU/MPU6050_DMP6_Teapot/MPU6050_DMP6_Teapot.ino
)

Sympton 1. digitalPinToInterrupt(INTERRUPT_PIN) error
digitalPinToInterrupt not supported ?

after commented digitalPinToInterrupt
Symptom 2. printfloatX error

/home/rkuo/Documents/Arduino/examples/libraries/MPU6050/src/MPU6050.cpp:3379:2:
note: in expansion of macro 'printfloatx'
  printfloatx("", Data[2], 5, 0, "\n");
  ^~~~~~~~~~~
exit status 1
Compilation error: exit status 1

xidameng, 2022-10-20:
Hi @rkuo2000 

Symptom1: digitalPinToInterrupt(INTERRUPT_PIN) error
`digitalPinToInterrupt` is indeed NOT supported on Ameba D as of now, but adding it is very simple so it would be added in the upcoming release

Symptom 2. printfloatX error because missing dtostrf
This is already fixed in the PR above, please expect this fix in the next release

Sympton 3. successfully upload to RTL8720DN, but hang after mpu6050 initialization
There are a few changes needed to make, but after that, the example works on my side, check my log below 

> #calibration_ok:[2:19:11] 
> 15:02:44.204 -> Initializing I2C devices...
> 15:02:44.204 -> Testing device connections...
> 15:02:44.204 -> MPU6050 connection successful
> 15:02:44.204 -> 
> 15:02:44.204 -> Send any character to begin DMP programming and demo: 
> 15:02:51.761 -> Initializing DMP...
> 15:02:52.372 -> >......>......
> 15:02:54.384 -> //           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
> 15:02:54.384 -> //OFFSETS    -6270,     483,    1638,     150,      70,      11
> 15:02:54.384 -> Enabling DMP...
> 15:02:54.384 -> Enabling interrupt detection (Arduino external interrupt 2)...
> 15:02:54.433 -> DMP ready! Waiting for first interrupt...
> 15:02:54.433 -> Error Invalid pin. 
> 15:02:54.433 -> ypr	0.01	-0.01	0.44
> 15:02:54.433 -> Error Invalid pin. 
> 15:02:54.433 -> ypr	0.01	-0.01	0.44
> 15:02:54.433 -> Error Invalid pin. 
> 15:02:54.433 -> ypr	0.01	-0.01	0.43
> 15:02:54.433 -> Error Invalid pin. 
> 15:02:54.480 -> ypr	0.01	-0.01	0.43
> 

PS: I didn't use your code, as it's heavily modified. The code I used is this one
https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino

> Look out to my PR for all the changes and bug fixes

xidameng, 2022-10-21:
Hi @rkuo2000 

Please refer to PR #139 for fix to your issues.

Also, there are 2 things you need to change on your side,

1.  On Arduino IDE, under `Tools` tab, look for `Standard Lib Enable:` menu, then choose `Arduino_STD_PRINTF` option. This would enable GCC standard lib such as <stdio.h> and <string> and etc. so as to solve the `Format not supported` error

2. On top of your example code, re-define `M_PI` to `PI` macro
`#define     M_PI   PI`

These 2 fixes are required to compile and run the example correctly, after these fix, the example works fine

M-ichae-l, 2022-10-26:
@xidameng has provide the solution, processed to closed the issue.

closed 2022-10-26 by M-ichae-l