Ameba-AIoT/ameba-arduino-d #133: RTL8720DN PS4 controller driver ?

rkuo2000, 2022-10-06:
Here's a PS4 controller driver for ESP32
[https://github.com/aed3/PS4-esp32](https://github.com/aed3/PS4-esp32)

For RTL8720DN, how to modify the following two ?
[ps4_l2cap.c](https://github.com/aed3/PS4-esp32/blob/master/src/ps4_l2cap.c)  
[ps4_spp.c](https://github.com/aed3/PS4-esp32/blob/master/src/ps4_spp.c)

xidameng, 2022-10-18:
Hi @rkuo2000 

This library is heavily dependent on the `esp-idf` bluetooth libraries, not on the unified Arduino Bluetooth API, thus supporting 
 this library means porting and re-writing almost the entirety of this library...  

Thus , this feature is currently not supported.

However, we do feel adding support for a PS controller is good idea for our users. Therefore, if you would help to initiate this effort, we would also love to provide supports along the way 😃

closed 2022-10-18 by xidameng

rkuo2000, 2022-10-18:
Hi, 
Yes, PS4 controller is very useful to remote control Robots, especially Drones (Quadcopter, ...).
It seems to me, ps4_spp.c and ps4_l2cap.c need to be created,
but I don't have the knowledge of your bluetooth stack to write those codes.