Ameba-AIoT/ameba-arduino-d #134: _rtl_sprintf: format not support!

bkrajendra, 2022-10-15:
I am using a simple sprintf in my code.
```
sprintf(path, "/input/post?node=%s&json={humidity:'%2.1f',temp: '%2.1f'}', 'apikey':%s}", cid, h, t, api_key);
```
what could be wrong here.?

xidameng, 2022-10-17:
Hi @bkrajendra ,

Thanks for your feedback, Ameba D by default uses custom `sprintf` implementation in its ROM, thus it doesn't support all `format specifier` such as `%lf` etc. This is so to save memory and prevent malicious attacks targetting RAM.

However, this SDK also supports standard lib implementation, we will add a easy-to-use option in the Arduino IDE drop down menu later, but for now, you can follow my steps below to fix your issue,

1. Locate the `Arduino.h` file on your PC, for my case, the path looks like this if you are using the **Latest Version**,

> %USERPROFILE%\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.4\cores\ambd

2. Uncomment the macro `Arduino_STD_PRINTF` and add another macro 

Nagivate to line 30, and uncomment this line, then add `#define STD_PRINTF` in the line above

xidameng, 2022-10-17:
This issue is already fixed in #136 and proceed to close to the issue

closed 2022-10-17 by xidameng