Ameba-AIoT/ameba-arduino-d #140: WPA2 Enterprise WIFI support

AQIBRIAZ, 2022-10-24:
Hi,

Can you let me know how to connect Enterprise wifi? is it supported or not?

Thanks and Regards

S10143806H, 2022-10-25:
Hi, 
The WPA2 function is supported at the Wi-Fi driver level. For testing WPA2 mode, you might need to configure your associated AP with the security of WPA2-PSK TKIP, WPA2-PSK AES, or WPA2-PSK TKIP&AES mixed mode. To ensure the WPA2 mode is working, make sure your PC and Ameba board connect to the same AP and perform a ping test.
Thanks

AQIBRIAZ, 2022-10-25:
Hi,

WPA2 personal is working fine. I am trying to connect to WPA2 Enterprise network. It required additional parameters and certificate .

Does this arduino sdk support the enterprise network ?

Regards

M-ichae-l, 2022-10-26:
@AQIBRIAZ , please refer to https://github.com/ambiot/ambd_sdk/releases/tag/v1.0.1, Arduino SDK not support EAP.

closed 2022-10-26 by M-ichae-l