Ameba-AIoT/ameba-arduino-d #143: Bring back two more ADCs on RTL8720DN officially

be1ieve, 2022-10-28:
Since new boards use LOG_UART on USB connector, PB1 and PB2 are available now.
Can we officially enable A0 and A1 and only mark old BW16 model as unusable on website?

PS: Calling pinMode() and analogRead() directly on PB1 and PB2 works already.
It's just like a hidden feature for those who read datasheet.

Aurical, 2022-11-01:
You could just directly uncomment `#define A0` and `#define A1` in the [variant.h file](https://github.com/ambiot/ambd_arduino/blob/dev/Arduino_package/hardware/variants/rtl8720dn_bw16/variant.h) for the board.

Selectively enabling it for the new board version would be much more complicated, it would require differentiating the two boards and creating a new set of pin definitions for the new board. This does not make much sense since the hardware for both boards are essentially the same.

M-ichae-l, 2022-11-07:
Close as there is no further discussion.

closed 2022-11-07 by M-ichae-l