Ameba-AIoT/ameba-arduino-d #156: Create an avr.io file for ambd_arduino

5t0n3r1pp3r, 2023-01-25:
this core needs to have an avr.io file.

A ton of universal arduino libraries rely on this file path / structure  and this is needed  for fast serial communications, RS485  and other uses.

If i can assist anyone in testing or developing one that has knowledge in that level of code please let me know.

S10143806H, 2023-02-20:
Hi @5t0n3r1pp3r,

Do you mean the avr/io.h under the file path below: `C:\Program Files (x86)\Arduino\hardware\tools\avr\avr\include\avr`?

Since io.h is under the AVR library but not the Arduino core. Could you please further illustrate why do you think it is necessary to have such file support?

Thank you.

5t0n3r1pp3r, 2023-02-20:
Biggest reason is compatibility, and a lot of esp32 libraries use it and would be easier to convert over libraries and add further support to the ambiot platform. 
additionally things like dmx simple and serial dmx libraries rely on this core file and other interrupt enabled/required libraries for communication.

this will make the core stronger in my opinion and open the path to more converted libraries

S10143806H, 2023-02-21:
Hi @5t0n3r1pp3r ,

Thanks for your explanations. It indeed sounds impressive.

Since I am new to this area. If it is possible, could you kindly please provide some examples of how esp32 libraries are using avr.io?

For the DMX libraries you mentioned, I have found this [file](https://github.com/TinkerKit/DmxMaster/blob/master/DmxMaster.cpp) included <avr/io.h>.

5t0n3r1pp3r, 2023-02-24:
I am a little new to the ways it uses it and only stumbled on this trying to convert certain libraries over to be used. 

I'm still looking into the issues but it seems the esp 32 and 8266 libraries have their own version of this file. I am trying to gather as much data as i can on it but i am working a lot this moth so it will be slow. 

However i am also coming to an assumption that this file may be a workaround or a "quick map" for certain core functions and most of this functionality may be implemented already, but that is just a thought from looking over stuff

S10143806H, 2023-02-27:
No problem. I will take a look at this file recently and meanwhile do feel free to leave your comments here if there are any new findings.
Again, thanks for your support :)
Happy Coding!

5t0n3r1pp3r, 2023-03-07:
sounds good there. I will look deeper into the calls for this file and see if it is actually needing another file for the DMX libraries and try to resolve the issue there. 

The main goal was to convert or create library for DMX to the ameba core so if there is any info on that or a recommended forum to check please let me know otherwise we can close this one.

closed 2023-03-08 by M-ichae-l