Ameba-AIoT/ameba-arduino-d #157: Cannot Set Serial with any secondary Qualifier like SERIAL_8N1 or SERIAL_8N2

5t0n3r1pp3r, 2023-01-25:
At this time i cannot seem to find a way to use the second option for serial communications with this board. this is preventing more custom Serial communications and pathways

M-ichae-l, 2023-01-26:
Can you upload pictures of hardware connection and the hardware you are using? 
And which amebad board you selected from the Arduino IDE?

5t0n3r1pp3r, 2023-02-06:
i am using to the BW16 with micro usb and connecting the proper pins to upload and have no issues there.
The problem is the Serial.begin(Speed,Config);
does not accept any config settings and it wants just Serial.begin(Speed);
therefore limiting the usage of this board to regular SERIAL_8N1 instead of the options listed at 
https://www.arduino.cc/reference/en/language/functions/communication/serial/begin/

M-ichae-l, 2023-02-07:
> i am using to the BW16 with micro usb and connecting the proper pins to upload and have no issues there. The problem is the Serial.begin(Speed,Config); does not accept any config settings and it [...]

Thanks for the reference. We will review and add this support.

M-ichae-l, 2023-02-07:
@5t0n3r1pp3r  Please refer to commit https://github.com/ambiot/ambd_arduino/commit/790c972f117e3cd71b58ea0214e077fd581df8c6

M-ichae-l, 2023-02-13:
close as no further issue.

closed 2023-02-13 by M-ichae-l