Ameba-AIoT/ameba-arduino-d #159: Error while compiling the Tensorflow lite examples for AMB23.

salmanfarisvp, 2023-02-05:
Hi, I added the tensorflow lite zip library from [here  ](https://github.com/ambiot/ambd_arduino/tree/dev/Arduino_zip_libraries) and getting errors while compiling the examples sketches on Arduino IDE. 

```
Arduino: 1.8.19 (Mac OS X), Board: "AMB23 (RTL8722DM_MINI), Disable"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/salman/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/salman/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/salman/Documents/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8722dm_mini:EraseFlash=Disable -vid-pid=0000_0000 -ide-version=10819 -build-path /var/folders/1x/5pzvf2b56794fn3nglyjr1xc0000gn/T/arduino_build_296593 -warnings=none -build-cache /var/folders/1x/5pzvf2b56794fn3nglyjr1xc0000gn/T/arduino_cache_387345 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_tools.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.1.0 -prefs=runtime.tools.ameba_d_tools-1.1.0.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.1.0 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -verbose /Users/salman/Documents/Arduino/libraries/Ameba_TensorFlowLite/examples/hello_world/hello_world.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/salman/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/salman/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/salman/Documents/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8722dm_mini:EraseFlash=Disable -vid-pid=0000_0000 -ide-version=10819 -build-path /var/folders/1x/5pzvf2b56794fn3nglyjr1xc0000gn/T/arduino_build_296593 -warnings=none -build-cache /var/folders/1x/5pzvf2b56794fn3nglyjr1xc0000gn/T/arduino_cache_387345 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_tools.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.1.0 -prefs=runtime.tools.ameba_d_tools-1.1.0.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_tools/1.1.0 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/Users/salman/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -verbose /Users/salman/Documents/Arduino/libraries/Ameba_TensorFlowLite/examples/hello_world/hello_world.ino
Using board 'ameba_rtl8722dm_mini' from platform in folder: /Users/salman/Library/Arduino15/packages/realtek/hardware/AmebaD/3.1.4
Using core 'ambd' from platform in folder: /Users/salman/Library/Arduino15/packages/realtek/hardware/AmebaD/3.1.4
Detecting libraries used...

Error while detecting libraries included by /Users/salman/Documents/Arduino/libraries/Ameba_TensorFlowLite/src/tensorflow/lite/core/api/flatbuffer_conversions.cpp
```

salmanfarisvp, 2023-02-05:
I was able to compile the examples after applying the patch https://github.com/ambiot/ambd_arduino/tree/dev/Ameba_misc/TensorFlowLite_patch .

M-ichae-l, 2023-02-06:
@salmanfarisvp  Is there any further issues?

M-ichae-l, 2023-02-14:
Closed as there is no further issue.

closed 2023-02-14 by M-ichae-l