Ameba-AIoT/ameba-arduino-d #169: RTL8722DM support parallel LCD

cold63, 2023-04-12:
Request for Arduino IDE to support parallel LCD functionality on RTL8722DM.

S10143806H, 2023-04-20:
Hi @cold63,
Could you please refer to the example [here](https://www.amebaiot.com/en/amebad-arduino-i2c-lcd/) and check whether this satisfied your request? 
Thank you.

cold63, 2023-04-22:
Sorry, I didn't express myself clearly. My request is to support the 16-bit parallel TFT LCD function, which is already supported in the standard SDK. If it is also supported on Arduino, Ameba will have a greater advantage over competing products in the same category.

S10143806H, 2023-04-24:
Hi, 

We currently already support the `ILI9341` 16-bit parallel TFT LCD driver in AmebaD Arduino SDK. 

May I know which type of driver you are aiming to be added to Arduino SDK? 

Thanks

M-ichae-l, 2023-04-24:
Any interface you are going to use for example I2C or SPI? It is better to have a example we can refer, like ILI9341

S10143806H, 2023-04-26:
Hi @cold63 ,

As per discussed with you in the private chat, you can use the external TFT_eSPI library on AmebaD to support desired features.

Please remember to comment out  Line 65~67 in `SPI.cpp` to avoid unwanted warnings while using BW16.
https://github.com/ambiot/ambd_arduino/blob/dff4ca4f404d0b7e621e7e7755a3867392fd0888/Arduino_package/hardware/libraries/SPI/src/SPI.cpp#L65-L67

If there are no further queries, please proceed to close this issue. 

Thank you~

cold63, 2023-04-26:
OK

closed 2023-04-26 by cold63