Ameba-AIoT/ameba-arduino-d #172: sscanf does not work when Arduino_STR_PRINTF is enabled

vladkozlov69, 2023-04-30:
The following fragment of code works correctly when Arduino_STR_PRINTF is disabled:
  
`    String mqttServer = "192.168.0.110";
    uint8_t ipAddress[4];
    mqttServer.trim();
    sscanf(mqttServer.c_str(), "%u.%u.%u.%u", &ipAddress[0], &ipAddress[1], &ipAddress[2], &ipAddress[3]);    
`
When I enable Arduino_STR_PRINTF, all bytes in ipAddress are 0x00 after sscanf. It it a known limitation or a bug?

vladkozlov69, 2023-05-01:
After some experiments I found that if I remove _strtol_r symbol from variants/rtl8720dn_bw16/linker_scripts/gcc/rom_symbol_ns.txt  then it works as expected.

M-ichae-l, 2023-05-05:
@vladkozlov69  Thanks for the solution. We will review and merge into SDK.

vladkozlov69, 2023-05-05:
@M-ichae-l I afraid we'll need two different linker scripts for Arduino_STD_PRINTF ON and OFF, otherwise it could lead to linker issues. There could be different offsets at least...

M-ichae-l, 2023-05-09:
@vladkozlov69  Ok, I think currently this is a walk around. Our team will run though and try to give a solution by using one linker script only. Thanks again!