Ameba-AIoT/ameba-arduino-d #173: Wifi Config with BLE

akycdi, 2023-05-03:
Hello, 
I have raised an issue in the forum with the same request can anyone suggest what can I do.

[forum](https://forum.amebaiot.com/t/configure-wifi-with-sending-ssid-and-pass-from-ble-i-have-to-configure-my-own-name-and-uuid/1837)

Thank you

S10143806H, 2023-05-05:
Hi @akycdi ,

We have reviewed your question but we think it is not clear enough. And our engineer has replied to you in [forum](https://forum.amebaiot.com/t/configure-wifi-with-sending-ssid-and-pass-from-ble-i-have-to-configure-my-own-name-and-uuid/1837) yesterday.

Kindly provide at least the following information in your reply since we currently support multiple SDKs and various boards:
1. Your ameba board 
2. Your SDK version (If you are downloading the SDK from GitHub, please specify the branch you are using)
3. Detailed steps to reproduce your issue

We seek your understanding and thanks for choosing Ameba. 😀

> Note: 
> To avoid redundancy, please directly contact our engineer on Forum and I will proceed to close this issue!

closed 2023-05-05 by S10143806H

akycdi, 2023-07-17:
Hi, Sorry for the late reply
My use case: I must use Wi-Fi in the 5G spectrum to transmit data from my Pico to my client's device. I must now set my own UUID and Name for my Ameba. In order to connect to the WiFi network that my client's device is linked to, I'm trying to send data to Ameba via BLE. I'm new to this ameba; if anyone can help, it would be great. I need to specify my own settings instead of the default ones (example is the same as BLEWIFIConfig).