Ameba-AIoT/ameba-arduino-d #187: Serial

huangbenlsl, 2023-07-21:
arduino3.1.6 
AMB23

Serial2‘s performance is inconsistent with Serial1  in high baud rate(>=2M)。

void threadB(void* pvParameters) {
  int recv_length;
  uint8_t recv_buff[1024];
  for (;;) {
     recv_length = Serial2.available();
     if(recv_length>0)
     {
       Serial2.readBytes(recv_buff,recv_length);
       if(loss_link_count < 200)
       {
          client.write(recv_buff,recv_length);
       }
       Serial.print("uart: ");
       Serial.print(recv_length);
       Serial.print("   ");
       Serial.println(recv_buff[0],HEX);
     }
     else
     {
      delay(5);
     }
  }
}

when packet is larger than 66 bytes,Serial1 can receive complete bytes,but Serial2(LP_UART) can't receive complete bytes。

M-ichae-l, 2023-07-24:
@huangbenlsl 
hi, we have tested Serial2 with Serial by using FTDI and CP2102 (USB to UART), with example demo “MultiSerial“ 2M baud rate.
There is no issue for FTDI but for CP2102 the max baud rate is 1.5M. Therefore, is issue may caused by the USB to UART chip.

huangbenlsl, 2023-07-24:
for test，the length of packet maybe larger than 120 bytes。
what puzzles me the most is WHY  SERIAL1 is OKAY in high baud rate(>=2M)。
Do you test serial1 with 2M&CP2102？

M-ichae-l, 2023-07-26:
for test result 
serial1 and serial2 can not support 2M with CP2102
serial1 and serial2 can support 2M with FTDI

huangbenlsl, 2023-07-28:
Thanks，usb2ttl module maybe causes this problem，but when connect amb23 with a STM32F405 bord,the problem still exists。

closed 2023-08-01 by M-ichae-l