Ameba-AIoT/ameba-arduino-d #193: How to rebuild `lib_arduino.a` ?

vshymanskyy, 2023-08-03:
I'm getting some unwanted prints from the wifi library, like this:
```log
interface 0 is initialized
interface 1 is initialized

Initializing WIFI ...
WIFI initialized
```

I found that I can disable them by commenting out:
```cpp
#define RTW_ENABLE_API_INFO
```

However this requires rebuilding `hardware/variants/common_libs/lib_arduino.a`

I'm using `arduino-cli`, if that matters

M-ichae-l, 2023-08-07:
hi @vshymanskyy ,
Normally we dont suggest to rebuild lib, but there is a way. 

You can rebuild all cpp/c files that inside the lib_arduino.a and output object files. Then make all o files to become the new lib_arduino.a. finally replace the old lib_arduino.a. 

The log itself is very useful for users, if you do not want the log you can use another UART as your log to avoid unwanted log. In this way you do not need to rebuild lib_arduino.a.

vshymanskyy, 2023-08-07:
Yes, thank you.
Is there a way I can redefine the UART that is used for the std printf ?

closed 2023-08-23 by M-ichae-l