Ameba-AIoT/ameba-arduino-d #195: Make use of DCT API?

vshymanskyy, 2023-08-04:
I need to store some settings in the form of key-value pairs in the NVM (Flash).
One option I found is to use DCT:
```cpp
// initial DCT
ret = dct_init(DCT_BEGIN_ADDR, MODULE_NUM, VARIABLE_NAME_SIZE, VARIABLE_VALUE_SIZE, 1, 0);
// register module
ret = dct_register_module(example_dct_module);
// open module
ret = dct_open_module(&dct_handle, example_dct_module);
```
However, looks like the DCT library is not linked, I'm getting:
```log
undefined reference to `dct_init(unsigned long, unsigned short, unsigned short, unsigned short, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
```

Another option is to use some kind of a File system (is there anything I can use out of the box?).

Also, to store any data in the flash, I need to define a dedicated region / partition in the flash memory.
What's the best practice to do that?

@xidameng please advise

vshymanskyy, 2023-08-04:
Weird, i see that `variants/common_libs/lib_dct.a` gets linked, but still looks like it lacks the relevant functions.

vshymanskyy, 2023-08-04:
I overlooked the "C" linkage. It builds successfully like this:
```
extern "C" {
  #include <dct.h>
}
```

I also had to add `"-I{ameba.sdkpath}/component/common/file_system/dct"` to the `compiler.ameba.c.include` in `platform.txt`. It would be great if it was included by default.

Didn't test It yet. I need to define a dedicated region / partition in the flash memory.

vshymanskyy, 2023-08-04:
So yeah, `DCT` actually works as a back-end for the Preferences library:
```log
/home/vshymanskyy/Arduino/Preferences_Test/Preferences_Test.ino:142:test_bytes:PASS
/home/vshymanskyy/Arduino/Preferences_Test/Preferences_Test.ino:144:test_zero_bytes:PASS
/home/vshymanskyy/Arduino/Preferences_Test/Preferences_Test.ino:146:test_utf8_key:PASS
/home/vshymanskyy/Arduino/Preferences_Test/Preferences_Test.ino:147:test_utf8_value:PASS
/home/vshymanskyy/Arduino/Preferences_Test/Preferences_Test.ino:148:test_remove_key:PASS
/home/vshymanskyy/Arduino/Preferences_Test/Preferences_Test.ino:149:test_clear_namespace:PASS
```

vshymanskyy, 2023-08-04:
So, there are 2 questions left:
- `"-I{ameba.sdkpath}/component/common/file_system/dct"` needs to be added to `compiler.ameba.c.include` in `platform.txt`.
- I need to define the flash region for the DCT. I noticed that DCT only requires the start address during initialization, the size is not specified explicitly. What's the best practice to configure the DCT?

S10143806H, 2023-08-07:
@vshymanskyy ,

The dct function is default wrapped in `lib_dct.a` under `/variants/common_libs/`, therefore, to compile dct successfully, you may add in `"-I{ameba.sdkpath}/component/common/file_system/dct"` in `platform.txt` Line 72.

For the DCT configuration, please refer to our standard sdk example located at `ambd_sdk\component\common\example\dct`. (https://github.com/ambiot/ambd_sdk/tree/dev/component/common/example/dct)

Thank you.

vshymanskyy, 2023-08-08:
The `Device Configuration Table` support was added and pushed to GitHub.
@S10143806H would you consider this:
1. Adding the required include path to the `platform.txt`
2. Adding the Preferneces library to your library package by default? https://github.com/vshymanskyy/Preferences

M-ichae-l, 2023-08-30:
@vshymanskyy 
Thanks for bring it out. As we have discovered, there is no example our any API for DCT demo at arduino level `Arduino_package/hardware/libraries`

But there is a `lib_dct.a ` at compilations process. So as you have provided the [Preference](https://github.com/vshymanskyy/Preferences). Why not just do the Pull Request, as a submodule of the SDK? 

- you can still maintain your own code at your repo
- it can be a part of the Ambd_arduino SDK

S10143806H, 2023-08-30:
> The `Device Configuration Table` support was added and pushed to GitHub. @S10143806H would you consider this:
> 
> 1. Adding the required include path to the `platform.txt`
> 2. Adding the Preferneces library to your library package by default? https://github.com/vshymanskyy/Preferences

Hi @vshymanskyy ,
Sorry for the late reply.
- Could you please define what is `Device Configuration Table` referring to? I could not find any related code in your repo. Thanks

1. As answered previously, the line including `lib_dct.a` have exited in `platform.txt`: https://github.com/ambiot/ambd_arduino/blob/8e9512a7ac1372b6cc123127da36324f8d0002a2/Arduino_package/hardware/platform.txt#L83C1-L83C1

3. Yes, you could go ahead with Pull Request your examples into our SDK upon verification. I have tested your examples but both of them seems could not compile successfully in AmebaD Arduino environment.
[image]
[image]

Therefore, maybe could you please clarify what are things that I could do to help you with your Pull Request contribution. And we could proceed to review your codes. Thank you.

vshymanskyy, 2023-08-31:
- DCT is short for `Device Configuration Table`, as far as I understand. The related parts of the library are highlighted here: https://github.com/ambiot/ambd_arduino/issues/195#issuecomment-1669594923
- Yes, `lib_dct.a` is linked, but `component/common/file_system/dct` is not included by default: https://github.com/ambiot/ambd_arduino/blob/8e9512a7ac1372b6cc123127da36324f8d0002a2/Arduino_package/hardware/platform.txt#L72
- As for the examples, looks like you're using the released version of `Preferences`, but you need to use the latest (unreleased) version from github. The default examples are using `Serial.printf` which is not available on `Amebad`. We can improve it either by changing the examples or by adding `Serial.printf` to your platform.

vshymanskyy, 2023-08-31:
Updated one of the examples so it doesn't rely on `Print::printf()`: https://github.com/vshymanskyy/Preferences/blob/main/examples/StartCounter/StartCounter.ino

vshymanskyy, 2023-10-11:
will get to it soon

vshymanskyy, 2023-11-16:
still not there;) sorry!

M-ichae-l, 2023-11-30:
@vshymanskyy 
Thanks for the patient, I have create a submoule of your repo main branch. #5a4b8b79ede9ec2502da29b34cd89e65749bfad6
You may want to review the update. If any update or debug I will help to merge.

vshymanskyy, 2023-12-14:
I'm alive!

M-ichae-l, 2023-12-15:
@vshymanskyy  
Could you review about the submodule system for the example you have ported?

vshymanskyy, 2023-12-30:
@M-ichae-l please elaborate, Didn't get your question

vshymanskyy, 2024-01-14:
@M-ichae-l ?

M-ichae-l, 2024-01-23:
@vshymanskyy 
Can you review the submodule here [https://github.com/ambiot/ambd_arduino/tree/dev/Arduino_package/hardware/libraries](https://github.com/ambiot/ambd_arduino/tree/dev/Arduino_package/hardware/libraries)
[image]

vshymanskyy, 2024-02-09:
ok, i should be able to verify it furing the weekend.
sorry for a delay, the project that used  it was put on hold, so i need to setup the environment again.
this holds back any progress

vshymanskyy, 2024-03-21:
Looks like everything is correct. Closing the ticket.
Thanks for all your help and support!

closed 2024-03-21 by vshymanskyy