Ameba-AIoT/ameba-arduino-d #198: mDNS service discovery

vladkozlov69, 2023-08-15:
Is there a way to perform mDNS service discovery using current API implementation?

M-ichae-l, 2023-08-17:
https://www.amebaiot.com/en/amebad-arduino-mdns-basic/

vladkozlov69, 2023-08-17:
Thanks @M-ichae-l , I have seen this article, here the Arduino IDE performs mDNS search to find Amebad board - but what I want is the opposite case: to perform mDNS lookup on Ameba side to find for example, MQTT server in local network. I know that I need to join Ameba network interface to multicast group to listen to the mDNS responses, and send mDNS queries to multicast address - but maybe there is a ready-to-use API in Realtek code?

vladkozlov69, 2023-08-20:
I have ended up creating a library for AmebaD, based on some ESP8266 code: https://github.com/vladkozlov69/RTL8720DN_mdns

closed 2023-08-20 by vladkozlov69