Ameba-AIoT/ameba-arduino-d #200: [MicroPython feature request] WiFi scan improve.

water5, 2023-09-04:
```python
wifi = WLAN(mode = WLAN.STA)
wifi.scan()
```

MicroPython Ameba port, WiFi scan result type is 'NoneType', that not easy to access, `list` type is good choice. and sometimes, `channel (or frequency)`, `hidden` item is useful, Ameba port have not that currently.

[https://docs.micropython.org/en/latest/library/network.html#network.AbstractNIC.scan](https://docs.micropython.org/en/latest/library/network.html#network.AbstractNIC.scan)

By the way, please enable [MicroPython repository](https://github.com/ambiot/micropython/tree/ameba)'s `Issues` feature.

M-ichae-l, 2023-09-05:
@water5  thanks for the first post. You may find the follow link for MicroPython issue page. 
https://github.com/ambiot/ambd_micropython/issues

water5, 2023-09-06:
@M-ichae-l Thanks.

closed 2023-09-06 by water5