Ameba-AIoT/ameba-arduino-d #218: Spelling mistakes (GitHub Action Spell Check)

designer2k2, 2024-02-09:
### Boards

-

### External Hardware

-

### Hardware Configuration

-

### Version

latest dev (checkout manually)

### IDE Name

-

### Operating System

-

### Auto Flash Mode

Disable

### Erase All Flash Memory (4MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

1500000

### Description

There are multiple spelling mistakes in this repo, mostly typos like:
./Arduino_package/hardware/cores/ambd/wifi_drv.cpp:18: modifed ==> modified

I would purpose a GitHub action "spell check" to automatically check the repo and list all.  This will then also check all future commits before they get merged.
Is there a interest into something like this here?

The spell check action: https://github.com/designer2k2/ambd_arduino/blob/designer2k2-patch-1/.github/workflows/spell-check.yml

Here is a example on how it runs: https://github.com/designer2k2/ambd_arduino/actions/runs/7846123899/job/21412100274#step:4:17
(this run reports about 1800 errors)

I can make a pull request with this action if wanted.

### Sketch

```cpp
-
```

### Error/Debug Message

```plain
-
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

M-ichae-l, 2024-02-13:
@designer2k2 
Thanks for pointed it out! Please refer to this commit #84a9d07235c2db534fc9cb3fddad09bada7ed6ad 
Under `Arduino_package/hardware`
- `cores` and `libraries` are API that should be checked.
- `system` and `variants` are code base that will remain un-checkable

designer2k2, 2024-02-13:
Hello @M-ichae-l.
the commit is a good start! Great to see it :-)

Some points from my side i would consider:
- ignore_words_list "celcius" is wrong, the temperature unit is written with a s: https://en.wikipedia.org/wiki/Celsius 
- ignore_words_list "Bellow" is wrong, below is correct in both cases as it means a value is underneath another one: https://www.oxfordlearnersdictionaries.com/definition/english/below

This currently results in 3 errors (2x Celsius, 1x below)

Further i suggest to remove the filter for specific files, that results in some other changes needed:

- add "ser" to the ignore_words_list  (due to the usage in SoftwareSerial)
- add "hart" to the ignore_words_list  (due to the Name in SoftwareSerial)

The fully changed spell_check.yml: https://github.com/designer2k2/ambd_arduino/blob/designer2k2-patch-1/.github/workflows/spell_check.yml

And a run with the modifications: https://github.com/designer2k2/ambd_arduino/actions/runs/7891851038/job/21537043921

M-ichae-l, 2024-02-14:
@designer2k2 
for  "ser" and "hart" are from third party files. 
- if there is no issue, will keep checking and maintaining
- if there is issue, it will be skipped to void further compatibility problem

designer2k2, 2024-02-14:
Thanks! 
https://github.com/ambiot/ambd_arduino/commit/e8df7125f9f0701eea4e57cb6fa07526cd1ced72 is closing this here 👍

closed 2024-02-14 by designer2k2