Ameba-AIoT/ameba-arduino-d #224: Documentation/instructions/support files for how to rebuild the libraries in "variants/common_libs" (source in system)

mickeysoft42, 2024-03-12:
### Hardware Environment

All supported by ambd package

### Related technical area

system libraries

### Is the feature request related to an un-fixed issue?

No

### Describe the ideal solution/reault/output.

Documentation that explains how to build the libraries found in variants/common_libs (e.g. lib_httpc.a) and any files like makefile to support the building of these libraries.

### Additional remarks

Unlike libraries in Arduino which are rebuilt every time you rebuild your ino project, the sources in the system directory are not rebuilt into the variants/common_libs directory, even if you make changes. It's unclear to me how these common_libs are being built and it would be useful for people that want to contribute with bug fixes/adding features.

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

M-ichae-l, 2024-03-12:
https://gcc.gnu.org/onlinedocs/gcc-4.9.4/gnat_ugn_unw/Building-a-library.html

alinneacsu, 2024-05-07:
Any update about this process ?

mickeysoft42, 2024-05-08:
> Any update for the documentation of this process ?

Unfortunately not. The only response so far is about how to build "a" library. I'm sure well intended, but most of us are well familiar with doing that, and it's not relevant to the original request/issue which is specific to the ambd package.

M-ichae-l, 2024-05-21:
Unfortunately there is no rebuild for the common libs. 
Add a third party lib is more easier for user to add extra features.