Ameba-AIoT/ameba-arduino-d #226: Error compiling for board Ai-Thinker BW16 (RTL8720DN)

KilimSerg, 2024-03-22:
### Boards

BW16 (RTL8720DN)

### External Hardware

ILI9341 display

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.8.13 and Arduino IDE 2.2.1

### Operating System

Win 10

### Auto Flash Mode

Disable

### Erase All Flash Memory (4MB)

Disable

### Standard Lib

Disable

### Upload Speed

1500000

### Description

Hello.
I don't speak English so I apologize in advance for any inaccuracies in Google Translate

I'm trying to replicate the “RTLWiFiAnalyzer” project on the BW16-kit (RTL8720DN) board.
 
[image]

and encountered a problem when compiling the program /
If there are no logical or syntax errors at the end of the compilation, the Arduino IDE throws an error that it cannot compile the program for a specific board

Arduino IDE 1.8.13
exit status 1
Error compiling for board Ai-Thinker BW16 (RTL8720DN).

As far as I understand from the compiler message, the problem lies in two libraries
“Arduino GFX library” and “lwip_netconf.h”.

Below is a listing of the program compilation:

In file included from c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:4:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:27:
c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:145:6: warning: "ESP_ARDUINO_VERSION_MAJOR" is not defined, evaluates to 0 [-Wundef]
 #if (ESP_ARDUINO_VERSION_MAJOR < 3)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:9:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:27:
c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32LCD8.h:5:6: warning: "ESP_ARDUINO_VERSION_MAJOR" is not defined, evaluates to 0 [-Wundef]
 #if (ESP_ARDUINO_VERSION_MAJOR < 3)
[... 93 lines cut ...]
 u32_t sys_now(void);
       ^~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/network/lwip/lwip_v2.0.2/src/include/lwip/opt.h:51:0,
                 from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/network/lwip/lwip_v2.0.2/src/include/lwip/errno.h:40,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\cerrno:42,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\ext\string_conversions.h:44,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\basic_string.h:5429,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\string:52,
                 from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\cores\ambd/Arduino.h:36,
                 from C:\Users\User\AppData\Local\Temp\arduino\sketches\7639B3E19B93B90426C91C194E6DEA76\sketch\RTLWiFiAnalyzer.ino.cpp:1:
C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/api/network/include/lwipopts.h:177:21: note: previous declaration with 'C++' linkage
 extern unsigned int sys_now(void);
                     ^~~~~~~
exit status 1
Error compiling for board Ai-Thinker BW16 (RTL8720DN).

C:\Program Files (x86)\Arduino\libraries\Arduino_GFX-master\src\databus\Arduino_HWSPI.cpp: In member function 'virtual bool Arduino_HWSPI::begin(int32_t, int8_t)':
C:\Program Files (x86)\Arduino\libraries\Arduino_GFX-master\src\databus\Arduino_HWSPI.cpp:184:17: error: 'SPI_MODE2' was not declared in this scope
     _dataMode = SPI_MODE2;
                 ^~~~~~~~~
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

M-ichae-l, 2024-06-12:
Could you try the new early release version 3.1.8-build20240610 for the SPI pin issue?

M-ichae-l, 2025-01-09:
Close as no further issue

closed 2025-01-09 by M-ichae-l