Ameba-AIoT/ameba-arduino-d #231: The SPI.h file error

cold63, 2024-05-21:
### Boards

BW16

### External Hardware

ili9341 TFT LCD

### Hardware Configuration

none

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windiws 10

### Auto Flash Mode

Enable

### Erase All Flash Memory (4MB)

Disable

### Standard Lib

Disable

### Upload Speed

1500000

### Description

My board is a BW16. When using the TFT_eSPI library, I encountered the following error:

\libraries\TFT_eSPI\TFT_eSPI.h:135:26: error: 'SPI_MODE0' was not declared in this scope

I modified the `SPI.h` file by changing `#define SPI_DATA_MODE0` to `SPI_MODE0` and line 58 from `SPI_DATA_MODE0` to `SPI_MODE0`. I also changed `SPI_DATA_MODE0` to `SPI_MODE0` in `SPI.c`.

After making these modifications, I was able to compile successfully and execute the program correctly.

Additionally, I discovered another issue: if I set `Tools / Standard Lib` to `Arduino_STD_PRINTF`, the following error occurs:
```
\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\stl_algobase.h:243:5: note: 'std::min'
min(const _Tp& __a, const _Tp& __b, _Compare __comp)
^~~

### Sketch

```cpp
#include <TFT_eSPI.h>
#include <SPI.h>
#include "sample.h"

TFT_eSPI tft = TFT_eSPI();

void setup() {
  // put your setup code here, to run once:
  tft.init();
  tft.setRotation(1);  // landscape

  tft.fillScreen(TFT_BLACK);

  // Swap the colour byte order when rendering
  tft.setSwapBytes(true);

  // Draw the icons
  tft.pushImage(0, 0, 320, 240, sample);

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Error/Debug Message

```plain
In file included from d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.cpp:16:0:
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.h:135:26: error: 'SPI_MODE0' was not declared in this scope
     #define TFT_SPI_MODE SPI_MODE0
                          ^
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.cpp:78:63: note: in expansion of macro 'TFT_SPI_MODE'
     spi.beginTransaction(SPISettings(SPI_FREQUENCY, MSBFIRST, TFT_SPI_MODE));
                                                               ^~~~~~~~~~~~
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::begin_nin_write()':
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.h:135:26: error: 'SPI_MODE0' was not declared in this scope
     #define TFT_SPI_MODE SPI_MODE0
                          ^
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.cpp:90:63: note: in expansion of macro 'TFT_SPI_MODE'
     spi.beginTransaction(SPISettings(SPI_FREQUENCY, MSBFIRST, TFT_SPI_MODE));
                                                               ^~~~~~~~~~~~
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.h:135:26: error: 'SPI_MODE0' was not declared in this scope
     #define TFT_SPI_MODE SPI_MODE0
                          ^
d:\Project\Arduino_lib\libraries\TFT_eSPI\TFT_eSPI.cpp:140:68: note: in expansion of macro 'TFT_SPI_MODE'
     spi.beginTransaction(SPISettings(SPI_READ_FREQUENCY, MSBFIRST, TFT_SPI_MODE));
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

pammyleong, 2024-06-11:
Hi @cold63 ,

You can try again with the new early release version 3.1.8-build20240610.

Thank you.

M-ichae-l, 2024-06-20:
close as no further discussion

closed 2024-06-20 by M-ichae-l