Ameba-AIoT/ameba-arduino-d #232: BW16 MQTT can't use ArduinoJSON.h

soyoung14541, 2024-05-30:
### Boards

BW16 

### External Hardware

NO

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino 2.3.2

### Operating System

Win 11

### Auto Flash Mode

Disable

### Erase All Flash Memory (4MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

1500000

### Description

Hello! 
First of all, my English expression skills are not very good, so I use Google Translate to explain my question. Please forgive me if there are any semantic problems.

I recently tried to use the BW16 development version to write MQTT code and wanted to use JSON format to process messages. But when I import ArduinoJSON.h, the system will report an error message, but the code before importing can be executed smoothly.
I tried to use ArduinoJSON.example -> StringExample alone. The sample program also failed to run after execution. The compilation could only be completed by turning on Standard Lib, but other problems would occur in the main program code.
I would like to ask how to deal with this. Thank you.

### Sketch

```cpp
#include <ArduinoJson.h>

#include <WiFi.h>
#include <PubSubClient.h>

#include <String>

char ssid[] = "DAXIN Biotechnology_2.4G";       // your network SSID (name)
char pass[] = "66483365";           // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                   // your network key Index number (needed only for WEP)

char server[] = "broker.emqx.io";      // the MQTT server of LASS
char clientId[17] = "";                    // client id for MQTT
char outTopic[20] = "UPsend"; // MQTT publish topic
String Sendpay = "";

WiFiClient wifiClient;
PubSubClient mqttClient(wifiClient);

void reconnectWiFi() {
    // attempt to connect to Wifi network:
    if (WiFi.status() != WL_CONNECTED) {
        Serial.print("Try connect to wifi ssid: ");
        Serial.println(ssid);
        while (WiFi.begin(ssid, pass) != WL_CONNECTED) {
            delay(1000);
        }
        Serial.println("Connected to wifi");
       
    }
}

void initializeMQTT() {
    byte mac[6];

    WiFi.macAddress(mac);
    
    Serial.print("MQTT client id:");
    Serial.println(clientId);
    Serial.print("MQTT topic:");
    Serial.println(outTopic);

    mqttClient.setServer(server, 1883);
}

void sendMQTT(String Sendpay) {
    char payload[Sendpay.length()];
    strcpy(payload, Sendpay.c_str());
       
    if (!mqttClient.connected()) {
        Serial.println("Attempting MQTT connection");
        mqttClient.connect(clientId);
    }

    if (mqttClient.connected()) {
        mqttClient.publish(outTopic, payload);
    }
}

void setup() {
    Serial.begin(115200);
    while (!Serial) {
        ;
    }

    reconnectWiFi();
    initializeMQTT();
}

void loop() {
    Sendpay = Sendpay+ "{\'temp\':' + (19+random(10)) + ', \'httpw\' : ' + 20 + ' }";
    
    reconnectWiFi();
    sendMQTT(Sendpay);
    mqttClient.loop();
    delay(1000);
    Sendpay = "";
}
```

### Error/Debug Message

```plain
In file included from c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\basic_ios.h:37:0,
                 from c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\ios:44,
                 from c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\ostream:38,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/JsonString.hpp:8,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/Adapters/JsonString.hpp:8,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/StringAdapters.hpp:8,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StringPool.hpp:11,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/ResourceManager.hpp:8,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/JsonVariantConst.hpp:10,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:9,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/JsonArray.hpp:7,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:29,
                 from c:\Users\henry60706\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\henry60706\Desktop\實習\Arduino\HUB 5168\LASS_Basic_MQTT\LASS_Basic_MQTT.ino:1:
c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\locale_facets.h:2565:44: error: macro "isspace" passed 2 arguments, but takes just 1
     isspace(_CharT __c, const locale& __loc)
                                            ^
c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\locale_facets.h:2571:44: error: macro "isprint" passed 2 arguments, but takes just 1
     isprint(_CharT __c, const locale& __loc)
                                            ^
c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\locale_facets.h:2583:44: error: macro "isupper" passed 2 arguments, but takes just 1
     isupper(_CharT __c, const locale& __loc)
                                            ^
c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\locale_facets.h:2589:44: error: macro "islower" passed 2 arguments, but takes just 1
     islower(_CharT __c, const locale& __loc)
                                            ^
c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\locale_facets.h:2595:44: error: macro "isalpha" passed 2 arguments, but takes just 1
     isalpha(_CharT __c, const locale& __loc)
                                            ^
c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\locale_facets.h:2601:44: error: macro "isdigit" passed 2 arguments, but takes just 1
     isdigit(_CharT __c, const locale& __loc)
                                            ^
c:\users\henry60706\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\locale_facets.h:2613:45: error: macro "isxdigit" passed 2 arguments, but takes just 1
     isxdigit(_CharT __c, const locale& __loc)
                                             ^
exit status 1

Compilation error: exit status 1
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

DeqingSun, 2025-04-30:
Do something like
```
#define ARDUINOJSON_ENABLE_STD_STREAM 0
#include <ArduinoJson.h>
```
https://arduinojson.org/v7/config/enable_std_stream/

DeqingSun, 2025-05-07:
Some update

I got the same error when using tensorflowlite. The error can be triggered as simple as

```
#include <iterator>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

git-78, 2025-09-07:
Same for me. The lack of real support for Ameba controller is worrying. Apart from this error, I've plenty when porting code from ESP32. I'm likely to stick to ESP32 and to forget the BW16